DreamSourceLab/DSView #227: DSView V1.0 DSLogic内部测试波形错误

359303267, 2019-09-10:
[image]  

[image]  

使用0.99版本没有问题，使用1.0版本波形就出问题。不过好像实际采样并没有错误，只是内部测试波形有问题。  
如果在0.99版本下装入固件，然后再在1.0版本下进行内部测试，波形就没有问题了。推断是新版的固件问题。

DreamSourceLab, 2019-09-11:
没发现有啥问题。新版本需要硬件重新上电初始化。
[image]

359303267, 2019-09-11:
@DreamSourceLab  
新版固件绝对有问题！替换0.99版本目录下的res/DSLogicPlus.bin文件就正常，使用新版DSLogicPlus.bin文件就出问题，不仅内部测试出错，实际捕捉到的波形也是错误的。

a-sv, 2019-09-13:
I have exactly the same problem. DSLogicPlus.bin from version 0.99 works correctly, but 1.00 version not.
[image]

DreamSourceLab, 2019-09-15:
像是个别设备能复现这个问题，可以试试下面这个修复的bin文件
[DSLogicPlus.zip](https://github.com/DreamSourceLab/DSView/files/3613609/DSLogicPlus.zip)

a-sv, 2019-09-15:
Im tested your new firmware by internal test and signal capture, it works fine.

359303267, 2019-09-15:
@DreamSourceLab  
这个固件没问题了，希望能尽快发布一版修复此问题的版本。

closed 2019-09-18 by DreamSourceLab