DreamSourceLab/DSView #228: Time markers sometimes overlap

milesfrain, 2019-09-10:
The time tick labels sometimes overlap on startup. This might be related to using a HiDPI display.
[image]

This is usually fixes itself when a user changes zoom levels with mouse scroll wheel.
[image]

Specifically, a user must zoom out one level beyond the startup view level.
Example:
```
Startup <- squish
Zoom out 1 level <- fixed
Zoom in 1 level <- still fixed, and always fixed
```
If a user zooms in, all of the zoomed-in levels will be fixed, but then the user will still encounter the issue when zooming back out to the startup level.
```
Startup <- squish
Zoom in n levels <- fixed
Zoom out n levels <- squished again
```

DreamSourceLab, 2019-09-11:
Any change of default system font size?

milesfrain, 2019-09-11:
Not that I'm aware of. This is on Ubuntu 18.04