DreamSourceLab/DSView #229: USB协议识别问题

Hifern, 2019-09-12:
使用DSLogic Plus检测USB协议目前看到有两个问题，一个是检测上电抓包异常，另一个是USB request无法解析。

第一个问题： USB D+，D－，GND接好线连入DSLogic Plus，设定好触发模式，初始默认断电，插入USB，触发响应。可以完整的抓完需要的波形，软件开始解析协议，但USB packet一层的协议会停在0%无法启动，而USB request则可跑完100%。
[image]
试过使用Bushound的Hot plug方式USB packet则可正常解析，但硬件直接去插拔USB口是肯定不行的。

第二个问题：USB request在检测协议时可以跑过100%但基本上没办法解析，我使用bushound和DSView V1.0同时抓同一组USB通信波形。
[image]
看USB packet解析的协议内容就很完整
[image]
但USB request识别不出任何一个请求命令。
[image]
而同样的信号在Bushound上是可以识别出来的，说明信号没问题，应该是DSView解析协议的问题。

DreamSourceLab, 2019-09-14:
@Hifern 
这些问题都是因为stack协议的添加顺序问题，USB signalling->USB packet->USB request这三个是依次叠加的关系，request是基于packet做解析，不是signalling。

Hifern, 2019-09-30:
非常感谢, 现在USB可以解析Request了, 但并不能说正常, 我这有一个抓取波形解析协议的文件, 但似乎出现了一次bulk in之后, 之前解析的所有request都被bulk覆盖掉了. 还请作者帮忙看看.
[USB Request异常.zip](https://github.com/DreamSourceLab/DSView/files/3669153/USB.Request.zip)