DreamSourceLab/DSView #232: how to uninstall DSview0.99(怎么卸载旧版本？)

ChenZhengwei20190741, 2019-09-18:
怎么卸载旧版本呀，最近想用这个新版本，安装完，还是旧版本，怎么卸载～

DreamSourceLab, 2019-09-18:
安装目录有uninst.exe可以卸载。另外，安装到相同目录也可以直接覆盖原来的版本。

ChenZhengwei20190741, 2019-09-19:
ubuntu版本的怎么卸载？卸载不干净呀，是不是设置了哪些环境变量？
然后在make的时候出现这个错误，是没安装Qt library吗？可是我0.99版本安装也没出现过问题？
chenzhengwei@madcow-OptiPlex-3050:~/esp/DSView-1.01/DSView$ make 
[  0%] Building CXX object CMakeFiles/DSView.dir/main.cpp.o
/home/chenzhengwei/esp/DSView-1.01/DSView/main.cpp: In function ‘int main(int, char**)’:
/home/chenzhengwei/esp/DSView-1.01/DSView/main.cpp:66:32: error: ‘AA_EnableHighDpiScaling’ is not a member of ‘Qt’
     QApplication::setAttribute(Qt::AA_EnableHighDpiScaling);
                                ^
CMakeFiles/DSView.dir/build.make:559: recipe for target 'CMakeFiles/DSView.dir/main.cpp.o' failed
make[2]: *** [CMakeFiles/DSView.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/DSView.dir/all' failed
make[1]: *** [CMakeFiles/DSView.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

closed 2019-09-19 by ChenZhengwei20190741