DreamSourceLab/DSView #243: Linux compile Error "AA_EnableHighDpiScaling"

jolinux, 2019-09-30:
Hello,

since version 1.00 DSLogic add 4K-Feature.

I compile DSView with Debian 7 (wheezy). Yes, it is a old OS-Version, but I need my Laptop for debug electrical devices. So I want not update my OS.

I see the error after make

`Scanning dependencies of target DSView
[ 30%] Building CXX object CMakeFiles/DSView.dir/main.cpp.o
/home/jo/dsview_1_01_fromgit/DSView/DSView/main.cpp: In function ‘int main(int, char**)’:
/home/jo/dsview_1_01_fromgit/DSView/DSView/main.cpp:66:32: error: ‘AA_EnableHighDpiScaling’ is not a member of ‘Qt’
     QApplication::setAttribute(Qt::AA_EnableHighDpiScaling);
                                ^
CMakeFiles/DSView.dir/build.make:621: recipe for target 'CMakeFiles/DSView.dir/main.cpp.o' failed
make[2]: *** [CMakeFiles/DSView.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/DSView.dir/all' failed
make[1]: *** [CMakeFiles/DSView.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2 `

The problem is "AA_EnableHighDpiScaling" .  A Qt 5.6. Function.

My OS
Linux version 3.16.0-7-amd64 (debian-kernel@lists.debian.org) (gcc version 4.9.2 (Debian 4.9.2-10+deb8u1) ) #1 SMP Debian 3.16.59-1 (2018-10-03)
with QT
Using Qt version 5.3.2 in /usr/lib/x86_64-linux-gnu

I change the source main.cpp  
(Line 65 and 68) to check the qt-version 

`int main(int argc, char *argv[])
{
	int ret = 0;
	struct sr_context *sr_ctx = NULL;
	const char *open_file = NULL;
#if QT_VERSION >= QT_VERSION_CHECK(5, 6, 0)
    QApplication::setAttribute(Qt::AA_EnableHighDpiScaling);
    QApplication::setAttribute(Qt::AA_UseHighDpiPixmaps);
#endif
    QApplication a(argc, argv); `

This help and make finish !!

Please import the patch.

Jörg

DreamSourceLab, 2019-10-03:
Got it.

DreamSourceLab, 2020-04-06:
Patched