DreamSourceLab/DSView #247: 200us/div 0轴显示， 500us/div时不显示

ncleijie, 2019-10-12:
115.2kbps的rs232信号的两组信号，相位差50ns，每隔100ms发送。 采集停止后， 在200us/div时两组信号可以正常显示；在500us/div时只有一组信号可以显示(用于边沿触发)，另外一组信号不能显示，一直为0. 异常显示图片不会添加。

ncleijie, 2019-10-12:
[示波器200us_500us波形显示异常2019-10-12.zip](https://github.com/DreamSourceLab/DSView/files/3720099/200us_500us.2019-10-12.zip)

ncleijie, 2019-10-12:
[image]

ncleijie, 2019-10-12:
[image]

DreamSourceLab, 2019-11-06:
@ncleijie 
我们下载了这个波形文件，但是没法复现问题。500us/div的时候，显示是正常的。
[image]

closed 2020-04-06 by DreamSourceLab