DreamSourceLab/DSView #254: 能否添加 eMMC 协议

sparkcore, 2019-11-06:
不知道是否可以添加 eMMC 协议呢？
谢谢

xtaens, 2022-05-06:
不知现在emmc的解析有了没

tony-wade, 2024-09-28:
我有做個陽春的能用，也能修改
[DSView - eMMC decoder](https://github.com/tony-wade/Reverse-Engineering/tree/main/DSView%20-%20eMMC%20decoder)

xtaens, 2025-02-19:
> 我有做個陽春的能用，也能修改 [DSView - eMMC decoder](https://github.com/tony-wade/Reverse-Engineering/tree/main/DSView%20-%20eMMC%20decoder)

兄弟你这个没有cmd线的选择啊，ds为啥和clk合到一起了呢，hs400下要用ds脚

tony-wade, 2025-02-19:
> > 我有做個陽春的能用，也能修改 [DSView - eMMC decoder](https://github.com/tony-wade/Reverse-Engineering/tree/main/DSView%20-%20eMMC%20decoder)
> 
> 兄弟你这个没有cmd线的选择啊，ds为啥和clk合到一起了呢，hs400下要用ds脚

所以是陽春的呀! ds讀的方式與clk無異，再拉一個decoder便是，ddr/sdr記得調。

ps. 當時上頭丟給我這機器要我一星期內報告所讀出資料與其分析，但光是eMMC的CMD和流程我就有得看了，看有沒有大神支援吧!