DreamSourceLab/DSView #257: Osc Width Auto Measure seems not to work

VaRu02, 2019-11-14:
See screenshot:
[image]

fgl27, 2019-11-14:
Seems a too high sample time...
Change the sample time in order to show more then one "off and on" time of yours signal...

You can manually change that at any time even after using the auto mode to setup the channel options.

Is known at least in all oscilloscope that I used that the auto mode will not be perfect for every signal. That is why you need to learn how to setup it up manually... There is a help on the app it self it opens the PDF that has all the info necessary of it featuring of the application check that.

VaRu02, 2019-11-14:
Well, there is only one "on and off" signal. I am using a debug pin to measure the duration of some processing...
Should I conclude that the feature was not meant for that use case?

VaRu02, 2019-11-14:
OK, I retriggered the processing and it measures the 2nd Width+ event, but it feels like something needs to be improved.
Also after a successful measurement, stopping the osc, the running again, it takes a few seconds before the previous measurement is cleared.

fgl27, 2019-11-14:
I thought it was a periodic signal...

I assume that those measure options only work well on periodic signals...

I don't know for sure, maybe is a possible to use that to measure that particularly single "on time"...
Lets see what **DSView** devs say...

On the meantime you can use the the mesure option and add cursors...
[image]

click with the right mouse key to add cursors and use the math option to precice get the value.

Is not as fast as if that other option worked but is a option that works...

As the oscilloscope work well with periodic signal (even on variable periodic signal), and yours is not that is a issue.The app can sync properly as the signal is so sporadic...

Maybe is possible to improve lets what for the project developers to take a look.

DreamSourceLab, 2019-11-15:
@VaRu02 
Current auto measurements can only calcurate the right pulse width when there are at least 2 pulses within the view window. We will try to improve this.
By the way, there are some issues when auto measure in roll mode. We are also working on this. So manual measurements is a beneficial supplement under this cases.