DreamSourceLab/DSView #258: Linking problem, libsigrokdecode4DSL.so: undefined reference to `Py*`

abougouffa, 2019-11-16:
Hi,

I'm trying to compile DSView on Arch Linux, I'm getting a linker error!

## CMake output:
```bash
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3") 
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Checking for modules 'libsigrok4DSL >= 0.2.0;libusb-1.0 >= 1.0.16;libzip >= 0.10;libsigrokdecode4DSL>=0.4.0'
--   Found libsigrok4DSL , version 0.2.0
--   Found libusb-1.0 , version 1.0.23
--   Found libzip , version 1.5.2
--   Found libsigrokdecode4DSL, version 0.6.0-git-fe09628
-- Using Qt5
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/lib64/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.42") found components:  filesystem system thread 
-- Found libusb-1.0:
--  - Includes: /usr/include
--  - Libraries: /usr/lib/libusb-1.0.so
-- Found LIBZIP: /usr/lib/libzip.so  
-- Found FFTW threads: /usr/lib/libfftw3f_threads.so
-- Found FFTW: /usr/lib/libfftw3.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hacko/.cache/yay/dsview-git/src/DSView/DSView
...
...
[ 99%] Building CXX object CMakeFiles/DSView.dir/qrc_language.cpp.o
[100%] Linking CXX executable DSView
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `_Py_Dealloc'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyLong_FromLong'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyModule_Create2'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyObject_SetAttrString'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyType_IsSubtype'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyErr_Fetch'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyList_Insert'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyType_GetFlags'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyEval_RestoreThread'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyDict_Size'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyUnicode_FromString'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyObject_HasAttrString'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyBytes_AsString'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyList_SetItem'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyErr_Clear'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyArg_ParseTupleAndKeywords'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyCallable_Check'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyLong_AsUnsignedLongLong'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyBytes_Size'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PySequence_Size'
[... 46 lines cut ...]
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyLong_AsLong'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyExc_Exception'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `_Py_TrueStruct'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyFloat_Type'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyExc_IndexError'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyFloat_AsDouble'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PySet_Pop'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyType_FromSpec'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyEval_SaveThread'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PySequence_Check'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `Py_IsInitialized'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyObject_CallMethod'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyObject_GetAttrString'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyModule_AddIntConstant'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyList_New'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyObject_Str'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyErr_Format'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libsigrokdecode4DSL.so: undefined reference to `PyTuple_Size'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/DSView.dir/build.make:2966: DSView] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/DSView.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

abougouffa, 2019-11-19:
I figured out why I got this error, I fixed that by adding PythonLibs to the link libraries in CMake, I tested that in my machine, it compiles and works fine.

closed 2019-11-19 by abougouffa

abougouffa, 2019-11-25:
@DreamSourceLab !!

I opened a pull request a week ago, can you check it and merge it please, now I can't compile the new version from your repo, I have to modify the `CMakeLists.txt` each time I want to compile.

In the pull request, I added some new decoders from `libsigrokdecode` _(a decoder per commit)_, can you check and merge them?

Regards

DreamSourceLab, 2019-11-26:
Merged.
Thanks.

PR #262 (2019-11-26): Fix #258, and add new decoders
# Fix #258
I has a compiling error as specified in #258, I fixed that by adding `PYTHON_LIBRARIES` to link libraries in CMake.

I noticed also that the order matters, **I have to** add `PYTHON_LIBRARIES` **to the end** of link libraries, (after `PKGDEPS_LIBRARIES`)

# Add some new decoders
I noticed that the `libsigrokdecode` codebase contains new decoders, so I added them to be used in DSView (as I need one of them to work with)
commit b018742e (2019-11-26): Merge pull request #262 from abougouffa/master

Fix #258, and add new decoders
commit d5ad3518 (2019-11-26): Add PythonLibs to link libraries in CMake, Fix #258