DreamSourceLab/DSView #267: Hardware Operation fail, please replug the USB, 硬件操作失败，请重新插拔...

doyenho, 2019-11-27:
常发生上述错误，插拔也无效
装置管理员正常

Always occur the error, unplug and plug did not fix it.
The device manager shows DSL device works.

DreamSourceLab, 2019-11-28:
这种问题基本都是因为USB连接不稳定造成的。
首先确认USB数据线没有问题，然后不要过hub或者机箱前面板USB端口，直接接主板上自带的USB端口。
确认硬件指示灯变成绿色之后再操作。

doyenho, 2019-11-28:
一直插在笔电上很多天，不用时Win 10睡眠，突然就发生问题(红灯无法变绿)。
有时Type C USB转180度再插; 有时换USB port插就正常了。
我也用Saleae，没遇过类似问题，二者都用WinUSB(Bulk Mode)，有重送机制。
装置管理员「停用装置」，再「启用装置」，能认到DSL，所以USB通讯应是正常的，也许是无法切到High Speed吧? Saleae只用Full Speed

DreamSourceLab, 2019-11-30:
睡眠的时候USB端口的供电可能会改变，需要重新插拔才能正常工作。

VladRTT, 2019-12-06:
[image]
Hi, I have the same problem. USB connection good, device shows in device manager as working fine.

closed 2020-05-12 by DreamSourceLab