DreamSourceLab/DSView #277: cannot make install libsigrokdecode4DSL

HanG-94, 2020-01-08:
when i try to install DSview/libsigrokdecode4DSL, encounter several Errors:
~/Software/DSView/libsigrokdecode4DSL$ sudo make install
 /bin/mkdir -p '/usr/local/lib'
 /bin/bash ./libtool --silent  --mode=install /usr/bin/install -c   libsigrokdecode4DSL.la '/usr/local/lib'
 /bin/mkdir -p '/usr/local/include/libsigrokdecode4DSL'
 /usr/bin/install -c -m 644 version.h '/usr/local/include/libsigrokdecode4DSL'
 /bin/mkdir -p '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 libsigrokdecode4DSL.pc '/usr/local/lib/pkgconfig'
 /bin/mkdir -p '/usr/local/include/libsigrokdecode4DSL'
 /usr/bin/install -c -m 644 libsigrokdecode.h '/usr/local/include/libsigrokdecode4DSL'
make  install-data-hook
/bin/mkdir -p /usr/local/share/libsigrokdecode4DSL/decoders
python3.7 ./tools/install-decoders \
	-i ./decoders -o /usr/local/share/libsigrokdecode4DSL/decoders
/bin/bash: python3.7: command not found
Makefile:1493: recipe for target 'install-decoders' failed
make[2]: *** [install-decoders] Error 127
Makefile:1401: recipe for target 'install-data-am' failed
make[1]: *** [install-data-am] Error 2
Makefile:1345: recipe for target 'install-am' failed
make: *** [install-am] Error 2

abougouffa, 2020-01-09:
I think the main problem is `python3.7: command not found`, you must install Python 3 and retry, maybe the installed Python version in your machine is 2, you have to install Python 3

HanG-94, 2020-01-10:
> I think the main problem is `python3.7: command not found`, you must install Python 3 and retry, maybe the installed Python version in your machine is 2, you have to install Python 3

thanks for answer, but i have already installed several Python versions, including Python3.7:
~/Software/DSView/libsigrokdecode4DSL$ python
python             python2.7-config   python3.6          python3.6m-config  python3.7m         python3m           
python2            python2-config     python3.6-config   python3.7          python3.7m-config  python3m-config    
python2.7          python3            python3.6m         python3.7-config   python3-config     python-config