DreamSourceLab/DSView #283: DSView v1.0.1: Cannot open a .dsl-file created by the same program.

Oleg10011001, 2020-01-29:
The problem occurs when you try to open a .dsl-file by double-clicking on it. When you try to open a .dsl file, two error messages appear. The first error message comes from the MinGW Runtime Assertion and reports the following "Assertion failed! Program: C: \ Program Files \ DSView \ DSView.exe File: ../DSView/pv/device/inputfile.cpp, Line 42 Expression: _input" and suggests choosing one of the options: Abort, Repeat, and Skip. The second error message arrives simultaneously with the previous one from the DSView program itself. In it, DSViev claims that the opened file is "Not a valid DSView data file". After that, DSView closes due to an error in its operation.

DreamSourceLab, 2020-01-30:
@Oleg10011001 
Please try to open in File->Open menu of DSView.

Oleg10011001, 2020-01-31:
I said above:
"Moreover, if I first open DSView and then try to open the same .dsl-file through the File -> Open menu, it will open normally without any problems."
The problem only occurs when you try to open a dsl-file by double-clicking on it ...

DreamSourceLab, 2020-05-12:
Fixed in v1.1.2