DreamSourceLab/DSView #284: Qi decoder is not working

ikey-ru, 2020-01-30:
What am I doing wrong?
[image]

DreamSourceLab, 2020-01-31:
@ikey-ru 
Please upload the dsl file.

ikey-ru, 2020-01-31:
[qi.dsl.zip](https://github.com/DreamSourceLab/DSView/files/4141780/qi.dsl.zip)

DreamSourceLab, 2020-03-05:
@ikey-ru 
The square frequency in your signal is about 170KHz.
It seems not a standard Qi signal, which should be about 2KHz.
Thanks.

ikey-ru, 2020-04-17:
That's right, this is the modulated signal of the wireless charger. 
The main 170 kHz signal is present there. 
I don't know how to separate the flies from the cutlets. :(

closed 2020-05-12 by DreamSourceLab

o0vvv0o, 2021-11-27:
Same question here