DreamSourceLab/DSView #285: Checkboxes not visible

cz7asm, 2020-02-19:
Just built 1.01 -- (2019-09-15) on my Linux machine and noticed problems with widget rendering - I cannot see any checkboxes. 

[image]

DreamSourceLab, 2020-02-20:
@cz7asm 
Could you provide us the distributor and version of your linux machine.
Thanks.

cz7asm, 2020-02-20:
I'm using Void Linux distro. However, I noticed that today the checkboxes are there. I would love to know what solved it. Whether just simply rebooting the machine or additional libraries introduced by other software I installed afterwards. Do you have any idea under what circumstances it can lead to Qt behaving like this?

closed 2020-05-12 by DreamSourceLab