DreamSourceLab/DSView #289: DSView V1.10 x64版本使用DSLogic Plus 无法使用，提示硬件错误

David-lau, 2020-03-11:


David-lau, 2020-03-11:
发现是和鼠标冲突了

khoahuynhld, 2020-03-17:
same issue, cant use on MACOS or Win10 too, downgrade to version 1.0.1 works OK.

Oleg10011001-2, 2020-03-23:
And what exactly is the problem expressed in? I tried this version of the program on my Windows 7 x64 and it seems that the program with DSLogic Plus works fine.

vivozhang, 2020-03-26:
> same issue, cant use on MACOS or Win10 too, downgrade to version 1.0.1 works OK.

I have the same issue, can use 1.1 on MAC or Windows 10, where can I download old version?

khoahuynhld, 2020-03-26:
> > same issue, cant use on MACOS or Win10 too, downgrade to version 1.0.1 works OK.
> 
> I have the same issue, can use 1.1 on MAC or Windows 10, where can I download old version?

how about restart the software 2 time without pressing any button, it helped me.

DreamSourceLab, 2020-04-06:
1. check driver in device manager
2. remove any hubs in DSLogic connection
3. check the led indicator of DSLogic turn green  before any operations

closed 2020-05-12 by DreamSourceLab