DreamSourceLab/DSView #305: DSView v1.1.1: DSL files that were created in older versions of this program do not open.

Oleg10011001-2, 2020-04-29:
DSView v1.1.1: DSL files that were created in older versions of this program do not open. When you try to open a DSL file created in an earlier version of the program, an error message is displayed that says: "Not a valid DSView data file."

DreamSourceLab, 2020-04-30:
@Oleg10011001-2 
Please upload the DSL file

Oleg10011001-2, 2020-05-01:
[1.zip](https://github.com/DreamSourceLab/DSView/files/4565431/1.zip)

DreamSourceLab, 2020-05-06:
@Oleg10011001-2 
We can open all of the three dsl files using DSView v1.1.1
[image]

Oleg10011001-2, 2020-05-06:
It seems that the indicated problem with opening the DSL file occurs when the name of one of the folders is specified in Russian in the path to the file.

DreamSourceLab, 2020-05-12:
Fixed in v1.1.2