DreamSourceLab/DSView #307: v1.1.1 Win10 64. "Failed to save data" when saving capture. DSLogic U3Pro32.

UK-Lab-LC, 2020-05-04:
I've started getting an error message "Failed to save data. Failed to create zip file. Please check write permission of this path." when a file save is about 40% complete. 

The folder has write permissions. I've successfully saved multiple .dsl files in there today but this one is not working, and the only thing I've changed is the channel selection for the capture. 

With the captures that saved successfully, I was using channels 1-16. Channels 17-32 had cables attached to the LA, but not to my device. 
I switched to using channels 17-32 then captured the very same signals from my device but now it will let the save begin and reach about 40% progress, hang the system for a couple of seconds then return the error message described above. 

Removing/reattaching the LA to my system didn't help, nor did closing/reopening DSView. 

I've unplugged the headers from channels 17-32 and moved them to channels 1-16, repeated the capture and successfully saved in to the same folder. So it's not an issue with what I'm capturing or where it's being saved, rather it seems like it's related to the channels being used.

DreamSourceLab, 2020-05-07:
@UK-Lab-LC 
We have tried to use channles 17-32, and then file->save successfully.
Could you tould us more options for your case?
such as sample duration, sample rate etcs.
Or you can upload the setting file (.dsc), which you can get from file->settings->store
Thanks.

Aleksandr-Kai, 2020-05-12:
It also does not save or open .dsc files for me (v.1.1.2)

DreamSourceLab, 2020-05-12:
@Aleksandr-Kai 
Please upload one of your dsl file created in version 0.98 with non-English file nams.
Thanks.

Aleksandr-Kai, 2020-05-13:
[абв.zip](https://github.com/DreamSourceLab/DSView/files/4620058/default.zip)

> 
> 
> @Aleksandr-Kai
> Please upload one of your dsl file created in version 0.98 with non-English file nams.
> Thanks.

DreamSourceLab, 2020-05-13:
> [абв.zip](https://github.com/DreamSourceLab/DSView/files/4620058/default.zip)
> 
> > @Aleksandr-Kai
> > Please upload one of your dsl file created in version 0.98 with non-English file nams.
> > Thanks.

Works well (v1.1.2)
[image]

Aleksandr-Kai, 2020-05-13:
[image]

Aleksandr-Kai, 2020-05-13:
in version 0.9.9 it works

[image]

DreamSourceLab, 2020-05-13:
@Aleksandr-Kai 
Which operation system do you use?
And which link do you use to download the DSView?

Aleksandr-Kai, 2020-05-13:
all downloads from https://www.dreamsourcelab.com/download/
old versions are also downloaded and stored on my disk
system is windows 7 32

DreamSourceLab, 2020-05-13:
@Aleksandr-Kai 
It should be caused by x86 version.
Thanks.