DreamSourceLab/DSView #313: Segfault when zooming large data.

lromor, 2020-06-06:
Hi, I'm getting the following stack trace when using DSLogic. 

```
Application: DSView (DSView), signal: Segmentation fault
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Current thread is 1 (Thread 0x7f8c45457780 (LWP 26204))]

Thread 6 (Thread 0x7f8c277fe700 (LWP 26256)):
#0  0x00007f8c4915faa0 in g_slice_alloc () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007f8c49160ee4 in g_slist_append () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007f8c498e450a in set_new_condition_list (args=<optimized out>, di=0x7f8c1c002170) at type_decoder.c:796
#3  Decoder_wait (self=<optimized out>, args=<optimized out>) at type_decoder.c:870
#4  0x00007f8c49487cc9 in _PyMethodDef_RawFastCallKeywords () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
#5  0x00007f8c4948880a in _PyMethodDescr_FastCallKeywords () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
#6  0x00007f8c4928ad5c in _PyEval_EvalFrameDefault () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
#7  0x00007f8c493b11e6 in _PyEval_EvalCodeWithName () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
#8  0x00007f8c4948830e in _PyFunction_FastCallDict () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
#9  0x00007f8c4948b20d in _PyObject_Call_Prepend () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
#10 0x00007f8c4948922b in _PyObject_FastCallDict () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
#11 0x00007f8c4948ad3e in PyObject_CallMethod () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
#12 0x00007f8c498e01e2 in di_thread (data=0x7f8c1c002170) at instance.c:1053
#13 di_thread (data=0x7f8c1c002170) at instance.c:1031
#14 0x00007f8c4916a415 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007f8c4a63afa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
#16 0x00007f8c483924cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 5 (Thread 0x7f8c27fff700 (LWP 26254)):
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007f8c4918bf9f in g_cond_wait () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007f8c498e1a1d in srd_inst_decode (di=0x7f8c1c002170, abs_start_samplenum=abs_start_samplenum@entry=5832704, abs_end_samplenum=abs_end_samplenum@entry=5849088, inbuf=inbuf@entry=0x7f8c1c001b00, inbuf_const=inbuf_const@entry=0x7f8c1c005ee0 "", inbuflen=inbuflen@entry=16384, error=0x7f8c27ffed30) at instance.c:1228
#3  0x00007f8c498de2d8 in srd_session_send (sess=sess@entry=0x7f8c1c003360, abs_start_samplenum=abs_start_samplenum@entry=5832704, abs_end_samplenum=5849088, inbuf=0x7f8c1c001b00, inbuf_const=0x7f8c1c005ee0 "", inbuflen=16384, error=0x7f8c27ffed30) at session.c:273
#4  0x0000560e83c28033 in pv::data::DecoderStack::decode_data (this=0x560e85fb0e80, decode_start=0, decode_end=10000383, session=0x7f8c1c003360) at /home/lromor/Downloads/DSVIEW/DSView-1.01/DSView/pv/data/decoderstack.cpp:510
#5  0x0000560e83c298a6 in pv::data::DecoderStack::decode_proc (this=0x560e85fb0e80) at /home/lromor/Downloads/DSVIEW/DSView-1.01/DSView/pv/data/decoderstack.cpp:586
#6  0x00007f8c4a668615 in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.67.0
#7  0x00007f8c4a63afa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
#8  0x00007f8c483924cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 4 (Thread 0x7f8c39016700 (LWP 26208)):
#0  futex_reltimed_wait_cancelable (private=0, reltime=0x7f8c39015a50, expected=0, futex_word=0x7f8c39015d60) at ../sysdeps/unix/sysv/linux/futex-internal.h:142
#1  __pthread_cond_wait_common (abstime=0x7f8c39015b10, mutex=0x7f8c39015d10, cond=0x7f8c39015d38) at pthread_cond_wait.c:533
#2  __pthread_cond_timedwait (cond=cond@entry=0x7f8c39015d38, mutex=mutex@entry=0x7f8c39015d10, abstime=abstime@entry=0x7f8c39015b10) at pthread_cond_wait.c:667
#3  0x0000560e83bb3431 in boost::condition_variable::do_wait_until (timeout=..., m=..., this=0x7f8c39015d10) at /usr/include/boost/thread/detail/platform_time.hpp:326
#4  boost::condition_variable::timed_wait<boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>, bool (*)()> (pred=<optimized out>, wait_duration=..., m=..., this=0x7f8c39015d10) at /usr/include/boost/thread/pthread/condition_variable_fwd.hpp:236
#5  boost::this_thread::sleep<boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l> > (rel_time=...) at /usr/include/boost/thread/pthread/thread_data.hpp:272
#6  0x0000560e83baaeaa in pv::SigSession::hotplug_proc(boost::function<void (QString)>) (this=0x560e853c01e8, error_handler=...) at /usr/include/boost/date_time/time_duration.hpp:259
#7  0x0000560e83baca45 in boost::_mfi::mf1<void, pv::SigSession, boost::function<void (QString)> >::operator()(pv::SigSession*, boost::function<void (QString)>) const (a1=..., p=<optimized out>, this=<optimized out>) at /usr/include/boost/function/function_base.hpp:698
#8  boost::_bi::list2<boost::_bi::value<pv::SigSession*>, boost::_bi::value<boost::function<void (QString)> > >::operator()<boost::_mfi::mf1<void, pv::SigSession, boost::function<void (QString)> >, boost::_bi::list0>(boost::_bi::type<void>, boost::_mfi::mf1<void, pv::SigSession, boost::function<void (QString)> >&, boost::_bi::list0&, int) (a=<synthetic pointer>..., f=..., this=<optimized out>) at /usr/include/boost/bind/bind.hpp:319
#9  boost::_bi::bind_t<void, boost::_mfi::mf1<void, pv::SigSession, boost::function<void (QString)> >, boost::_bi::list2<boost::_bi::value<pv::SigSession*>, boost::_bi::value<boost::function<void (QString)> > > >::operator()() (this=<optimized out>) at /usr/include/boost/bind/bind.hpp:1294
#10 boost::detail::thread_data<boost::_bi::bind_t<void, boost::_mfi::mf1<void, pv::SigSession, boost::function<void (QString)> >, boost::_bi::list2<boost::_bi::value<pv::SigSession*>, boost::_bi::value<boost::function<void (QString)> > > > >::run() (this=<optimized out>) at /usr/include/boost/thread/detail/thread.hpp:117
#11 0x00007f8c4a668615 in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.67.0
#12 0x00007f8c4a63afa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
#13 0x00007f8c483924cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 3 (Thread 0x7f8c39ffd700 (LWP 26207)):
#0  0x00007f8c48387819 in __GI___poll (fds=0x7f8c34004a10, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007f8c49142136 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007f8c4914225c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007f8c488b3743 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007f8c4886115b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007f8c486b0e76 in QThread::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007f8c44eb9545 in ?? () from /lib/x86_64-linux-gnu/libQt5DBus.so.5
#7  0x00007f8c486baa67 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007f8c4a63afa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
#9  0x00007f8c483924cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 2 (Thread 0x7f8c3ae89700 (LWP 26206)):
#0  0x00007f8c48387819 in __GI___poll (fds=0x7f8c3ae88e80, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007f8c49dacfbf in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
#2  0x00007f8c4a63afa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
#3  0x00007f8c483924cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
[... 23 lines cut ...]
#26 0x00007f8c4a14d581 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007f8c4a14c60c in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007f8c4a14d581 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x00007f8c4a14c60c in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
[... cut ...]
#43 0x00007f8c488b3727 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#44 0x00007f8c45017401 in ?? () from /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#45 0x00007f8c4886115b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#46 0x00007f8c48869132 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#47 0x0000560e83b9d2e3 in main (argc=<optimized out>, argv=<optimized out>) at /home/lromor/Downloads/DSVIEW/DSView-1.01/DSView/main.cpp:164
[Inferior 1 (process 26204) detached]

```

DreamSourceLab, 2020-06-07:
@lromor 
Could you provide the version and enviroment informations?
if possible, please also upload the dsl file for reproducing this issue?
Thanks.

lromor, 2020-06-07:
DSView 1.0.1, 
Linux 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2 (2020-04-29) x86_64 GNU/Linux

Anyways, I really appreciate your product, solid, and very intuitive interface. Really appreciated! 
The first issue I opened was a complaint. Definitely changed my mind!

lromor, 2020-06-07:
Once I manage to reproduce it, I will send you back the DSL file :)