DreamSourceLab/DSView #326: Failed to Connect to the PC with v1.1.2 (shows a red led always)

jagdeepsingh02, 2020-07-08:
Hello guys i have dslogic analyzer from 4/5 years and it is not working anymore with the latest version of Dsview, i have v0.9.9 on another pc (with latest build of windows 10) and it works fine there.
- I disabled Driver signature enforcement before installing the latest version and tried installing manually the winusb drivers manually but it just doesn't work the led always remains RED.
please help guys thanks

EsaJaatinen, 2020-07-09:
Hi

I had the same issue. The fix was to connect the USB-C plug other way around.

Kind regards, Esa

jagdeepsingh02, 2020-07-10:
> Hi
> 
> I had the same issue. The fix was to connect the USB-C plug other way around.
> 
> Kind regards, Esa

Thanks for answering sir, i tried as you suggested but it doesn't work...

nurikk, 2020-07-29:
having same issue on mac, if I launch app from console I'm able to see this logs:
```log
⋊> /A/D/C/MacOS ./DSView                                                                                                                                                                                                              20:31:17
srd: Failed to load decoder clear.bat: import by name failed:ModuleNotFoundError:No module named 'clear'.
Protocol decoder enabled!

Starting a hotplug thread...

DreamSourceLab Hardware Attached!

DreamSourceLab hardware attached!
sr: DSL Hardware: dsl_dev_open: Unable to claim interface: LIBUSB_ERROR_NOT_FOUND.
```

nurikk, 2020-07-29:
Logs with higher debug level
```log
DreamSourceLab hardware detached!
sr: hwdriver: Scan of 'virtual-demo' found 1 devices.
sr: DSL Hardware: DSLogic: Closing device 0 on 20.33 interface 0.
sr: hwdriver: Scan of 'DSLogic' found 0 devices.
sr: hwdriver: Scan of 'DSCope' found 0 devices.
DreamSourceLab Hardware Attached!

DreamSourceLab hardware attached!
sr: hwdriver: Scan of 'virtual-demo' found 1 devices.
sr: DSL Hardware: Found an DSLogic device.
sr: hwdriver: Scan of 'DSLogic' found 1 devices.
sr: hwdriver: Scan of 'DSCope' found 0 devices.
sr: DSL Hardware: dsl_dev_open: Firmware upload was not needed.
sr: DSL Hardware: Opened device 0 on 20.35, interface 0, firmware 2.0.
sr: DSL Hardware: dsl_dev_open: Unable to claim interface: LIBUSB_ERROR_NOT_FOUND.
sr: DSL Hardware: config_set: setting Signal Max Height to 0
sr: DSL Hardware: config_set: setting pattern to 2
sr: DSL Hardware: config_set: setting channel mode to 20
sr: DSL Hardware: config_set: setting filter to 0
sr: DSL Hardware: config_set: setting DSO Trigger Slope to 0
sr: DSL Hardware: config_set: setting DSO Trigger Source to 0
sr: DSL Hardware: config_set: setting DSO Trigger Source to 0
sr: DSL Hardware: config_set: setting Trigger Holdoff Time to 0
sr: DSL Hardware: config_set: setting Trigger Margin to 8
```

DreamSourceLab, 2020-08-02:
@jagdeepsingh02 
Sorry, only v0.99 and before version support this obsolete model.

DreamSourceLab, 2020-08-02:
@nurikk 
What's your hardware version or device name? 
It works with v0.99 but not with v1.12?

jagdeepsingh02, 2020-08-02:
> @jagdeepsingh02
> Sorry, only v0.99 and before version support this obsolete model.

so there is no way to make it work with new versions of software?
and i would like to ask if is it possible to put online compiled installation files for 0.99, i can't find them anywhere...
thanks in advance

nurikk, 2020-08-02:
> @nurikk
> What's your hardware version or device name?
> It works with v0.99 but not with v1.12?

I'm using dslogic plus version 
It's works with 1.12 but I have to reconnect it many times (1 out of 20-30 times)
Status led is always red
I filmed video with this issue 

 https://drive.google.com/drive/folders/1HI-wWYmeu55l0Pl9iKUZaVj_EIQD1u8m

DreamSourceLab, 2020-08-04:
@jagdeepsingh02 
https://www.dreamsourcelab.com/software/DSView_v0.9.9_x64_setup.exe

fvonkistowski, 2020-09-08:
@jagdeepsingh02 I am quite late for the party. Had the same issue, for both DSView 1.12 and DSView 9.9 as suggested. What finally fixed it, was using a non-charging USB port on my computer. It does not make to much sense for me, as the protocols are the same, but I tried it on two different computers.

jauming, 2022-01-10:
0.9.9 works for my old dslogic,thx