DreamSourceLab/DSView #330: Trigger position is can't move

lengyechen, 2020-07-27:
I used DSLogic Plus, I need to adjust the Trigger position , but can't move , the position is always 1% .

DreamSourceLab, 2020-07-28:
@lengyechen 
Use buffer mode.
Trigger can only be set within hardware memory depth.

lengyechen, 2020-07-28:
[image]
hello,
    Thanks for your reply, I already used buffer mode,  it always move the trigger position. As attached.
    What should I pay attention ?

Best Regards,
Lengye

DreamSourceLab, 2020-07-29:
@lengyechen 
Under buffer mode, you can set the trigger position, and then restart your capture.
[image]

orlamartins, 2020-08-10:
Hi, I am having the same issue with my DSLogic device.
[image]
Is there a workaround for that?
Thx

WGraupner, 2020-08-14:
I have the same issue with older SW-version as well as after update to V1.1.2. for hardware DSLogicBasic.
Inspite of buffer mode and placing trigger position greater than 1 %, the trigger does not move.
I desperatly look for a hint /  solution.
Regards Winfried

DreamSourceLab, 2020-08-16:
@orlamartins @WGraupner 
The trigger position will be at the setting position only when trigger event happend after setting position.
For example, if the first sample hit the trigger, the trigger position must be 0, because there are no any samples before the first sample.

orlamartins, 2020-08-18:
> @orlamartins @WGraupner
> The trigger position will be at the setting position only when trigger event happend after setting position.
> For example, if the first sample hit the trigger, the trigger position must be 0, because there are no any samples before the first sample.

I see, that makes sense. Many thanks for replying it.

skiphansen, 2022-07-18:
I have a similar issue here with version 1.2.0 built from today's (July 18, 2022) git. I can set the trigger to about 30% but that's it. I really need 99% to capture the issue I'm trying to debug.

My guess is that this is a GUI scrolling issue.

[image]

DreamSourceLab, 2022-07-20:
@skiphansen 
You use RLE compression.
Trigger postion can only be set within hardware memory depth.
If you need 99%, please unselect RLE mode.

skiphansen, 2022-07-20:
@DreamSourceLab Thanks for the explanation and great product !    It's a life saver.