DreamSourceLab/DSView #332: DSlogic Failed to Connect to the MAC with v1.1.2 (shows a red led always)

nurikk, 2020-08-12:
Hi guys, my brand new DSLogic Plus has issue connecting to my MAC (10.15.5),

`
⋊> /A/D/C/MacOS ./DSView                                                                                                                                                                                                              20:31:17
srd: Failed to load decoder clear.bat: import by name failed:ModuleNotFoundError:No module named 'clear'.
Protocol decoder enabled!

Starting a hotplug thread...

DreamSourceLab Hardware Attached!

DreamSourceLab hardware attached!
sr: DSL Hardware: dsl_dev_open: Unable to claim interface: LIBUSB_ERROR_NOT_FOUND.
`

More verbose log
`
DreamSourceLab hardware detached!
sr: hwdriver: Scan of 'virtual-demo' found 1 devices.
sr: DSL Hardware: DSLogic: Closing device 0 on 20.33 interface 0.
sr: hwdriver: Scan of 'DSLogic' found 0 devices.
sr: hwdriver: Scan of 'DSCope' found 0 devices.
DreamSourceLab Hardware Attached!

DreamSourceLab hardware attached!
sr: hwdriver: Scan of 'virtual-demo' found 1 devices.
sr: DSL Hardware: Found an DSLogic device.
sr: hwdriver: Scan of 'DSLogic' found 1 devices.
sr: hwdriver: Scan of 'DSCope' found 0 devices.
sr: DSL Hardware: dsl_dev_open: Firmware upload was not needed.
sr: DSL Hardware: Opened device 0 on 20.35, interface 0, firmware 2.0.
sr: DSL Hardware: dsl_dev_open: Unable to claim interface: LIBUSB_ERROR_NOT_FOUND.
sr: DSL Hardware: config_set: setting Signal Max Height to 0
sr: DSL Hardware: config_set: setting pattern to 2
sr: DSL Hardware: config_set: setting channel mode to 20
sr: DSL Hardware: config_set: setting filter to 0
sr: DSL Hardware: config_set: setting DSO Trigger Slope to 0
sr: DSL Hardware: config_set: setting DSO Trigger Source to 0
sr: DSL Hardware: config_set: setting DSO Trigger Source to 0
sr: DSL Hardware: config_set: setting Trigger Holdoff Time to 0
sr: DSL Hardware: config_set: setting Trigger Margin to 8
`

Status led is always red.
But some times like 1 attempt from 20-30 it connects successfully, I tested same DSL on my windows laptop, with it connects flawlesly

Here are the videos with issue
https://drive.google.com/drive/folders/1HI-wWYmeu55l0Pl9iKUZaVj_EIQD1u8m

ClassicOldSong, 2020-12-26:
Same problem here on USB A cables.
Whenever the device is connected to the USB, DSView wont load up. As soon as I disconnect the device, it opens. Then I plug the device back in, it wont start capturing.

fklement, 2020-12-26:
Having the exact same problem as @nurikk and @ClassicOldSong. Did you found a solution yet?  
I already tried different connection settings but non are working.

magicmicros, 2021-01-02:
Will there be a solution to this anytime???

iopq1, 2021-01-28:
hello my dear friends. i faced the same problem. decision very simple - you need  choose "Demo device" and then "DSLogic"

[image]

ClassicOldSong, 2021-01-28:
Choosing demo mode then you are getting nothing real

iopq1, 2021-01-28:
oh. its obviously. first - select demo device. second - dslogic. good working

https://user-images.githubusercontent.com/76855313/106081963-e5642b80-612a-11eb-9518-f3b64e0ec2fe.mov

ClassicOldSong, 2021-01-28:
Tried multiple times, no use. It’s still complains about connection.

iopq1, 2021-01-28:
you device switch red led on green led?

ClassicOldSong, 2021-01-28:
It stays red.

iopq1, 2021-01-28:
apparently my case unique. may be your hardware problem? windows pc good working?

ClassicOldSong, 2021-01-28:
Tried on Linux, working perfectly fine.

iopq1, 2021-01-28:
im sorry that my help useless. may be somebody can help

nurikk, 2021-01-28:
Btw, I've tried my dslogic with sigrok, and getting same issue

iopq1, 2021-01-28:
how install sigrok on macos 11.1? 
im get mistake -

`imac@iMac-iMac ~ % brew install sigrok-cli
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
mpdecimal
==> Updated Casks
Updated 65 casks.

==> Installing sigrok-cli from rene-dev/sigrok
==> Downloading http://sigrok.org/download/source/libsigrokdecode/libsigrokdecod
Already downloaded: /Users/imac/Library/Caches/Homebrew/downloads/7ad39221afeb178fe5158b44a419f49c85fe31fc70a97358dfd59c3e84fe8e04--libsigrokdecode-0.5.0.tar.gz
==> Downloading http://sigrok.org/download/source/sigrok-cli/sigrok-cli-0.7.0.ta
Already downloaded: /Users/imac/Library/Caches/Homebrew/downloads/a1da1f03260b6f0cf164035d9b3bd0ab27744521de7681183938a7e316cc66e3--sigrok-cli-0.7.0.tar.gz
==> Installing dependencies for rene-dev/sigrok/sigrok-cli: rene-dev/sigrok/libsigrokdecode
==> Installing rene-dev/sigrok/sigrok-cli dependency: rene-dev/sigrok/libsi
==> ./configure --prefix=/usr/local/Cellar/libsigrokdecode/0.5.0
==> make install
Last 15 lines from /Users/imac/Library/Logs/Homebrew/libsigrokdecode/02.make:
      _get_channels in decoder.o
      _decoder_free in decoder.o
      ...
  "__Py_FalseStruct", referenced from:
      _Decoder_has_channel in type_decoder.o
  "__Py_NoneStruct", referenced from:
      _srd_decoder_doc_get in decoder.o
      _srd_inst_start in instance.o
      _Decoder_put in type_decoder.o
      _Decoder_wait in type_decoder.o
  "__Py_TrueStruct", referenced from:
      _Decoder_has_channel in type_decoder.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [libsigrokdecode.la] Error 1

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/rene-dev/homebrew-sigrok/issues

These open issues may also help:
building libsigrokdecode with "brew" on macOS broken https://github.com/rene-dev/homebrew-sigrok/issues/23
imac@iMac-iMac ~ % `

nurikk, 2021-01-28:
> how install sigrok on macos 11.1?
> i get mistake -
> 
> `imac@iMac-iMac ~ % brew install sigrok-cli
> Updating Homebrew...
> ==> Auto-updated Homebrew!
> Updated 2 taps (homebrew/core and homebrew/cask).
> ==> New Formulae
> mpdecimal
> ==> Updated Casks
> Updated 65 casks.
> 
> ==> Installing sigrok-cli from rene-dev/sigrok
> ==> Downloading http://sigrok.org/download/source/libsigrokdecode/libsigrokdecod
> Already downloaded: /Users/imac/Library/Caches/Homebrew/downloads/7ad39221afeb178fe5158b44a419f49c85fe31fc70a97358dfd59c3e84fe8e04--libsigrokdecode-0.5.0.tar.gz
> ==> Downloading http://sigrok.org/download/source/sigrok-cli/sigrok-cli-0.7.0.ta
> Already downloaded: /Users/imac/Library/Caches/Homebrew/downloads/a1da1f03260b6f0cf164035d9b3bd0ab27744521de7681183938a7e316cc66e3--sigrok-cli-0.7.0.tar.gz
> ==> Installing dependencies for rene-dev/sigrok/sigrok-cli: rene-dev/sigrok/libsigrokdecode
> ==> Installing rene-dev/sigrok/sigrok-cli dependency: rene-dev/sigrok/libsi
> ==> ./configure --prefix=/usr/local/Cellar/libsigrokdecode/0.5.0
> ==> make install
> Last 15 lines from /Users/imac/Library/Logs/Homebrew/libsigrokdecode/02.make:
> _get_channels in decoder.o
> _decoder_free in decoder.o
> ...
> "__Py_FalseStruct", referenced from:
> _Decoder_has_channel in type_decoder.o
> "__Py_NoneStruct", referenced from:
> _srd_decoder_doc_get in decoder.o
> _srd_inst_start in instance.o
> _Decoder_put in type_decoder.o
> _Decoder_wait in type_decoder.o
> "__Py_TrueStruct", referenced from:
> _Decoder_has_channel in type_decoder.o
> ld: symbol(s) not found for architecture x86_64
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> make: *** [libsigrokdecode.la] Error 1
> 
> If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
> https://github.com/rene-dev/homebrew-sigrok/issues
> 
> These open issues may also help:
> building libsigrokdecode with "brew" on macOS broken [rene-dev/homebrew-sigrok#23](https://github.com/rene-dev/homebrew-sigrok/issues/23)
> imac@iMac-iMac ~ % `

Download binaries https://sigrok.org/wiki/Downloads

iopq1, 2021-01-28:
i do it. but app not found my device -
[image]

iopq1, 2021-02-14:
i need help how connected dslogic to pulseview. in dsview app not working magic mouse

LordRafa, 2021-02-15:
Hello I have a similar issue on Linux using qemu/libvirt. If I try to redirect this device using a spice channel, I get LIBUSB_ERROR_NOT_FOUND perhaps there is something nasty going on with libusb and the device itself?

LianYangCn, 2021-02-27:
I got the same issue
Tried on windows working perfectly fine on same PC

LianYangCn, 2021-02-27:
same as #320

McKlaud76, 2021-04-13:
Same here with macOS 10.14.6, the highest version of DSView that recognises DSLogic Plus from 2017 is 0.99. Any newer cannot recognise and communicate with the unit, LED in red permanently.

nurikk, 2021-04-23:
Updating to mac os 11.2.2 didn't help

LianYangCn, 2021-05-13:
Hi gays，

For me, replacing a USB 3.0 port can work normally

Camfix, 2021-11-08:
Hi fellows,

I struggled in the same mud ... with Dslogic Basic whatever... on my Mac.
- Pulseview did not recognize the logic analyzer at all, same the Mac desktop. Had same window in Pulseview as iopq1  showed us above. 
- Installed Dsview v. 1.12
- Clicked some buttons to see what happens, read the manual for Dsview. "Start" button caused a fault message (not that again ... I haven't even started to capture data) and turned to grey afterwards (I love that!).  I recognized Dsview shows the proper device name after reading some more pages in the manual, which I have not switched to ... (magic!), but gadget still shows a red LED. The worst you can do: Unplug it straightly. I did. Success! Green LED lit up and "Start" button green again in the application. 
- Finally: The installation of Dsview and connecting the hardware needs a third line in the manual: Unplug your device for a moment. ;-)

Will see, if all the capturing etc. works. I am curious.
Keep on going!

 Andreas

iopq1, 2021-11-09:
hello. i have converted my dslogic to dslogicplus. now my device wors with pulseview. dsview i isn't like because what magic mouse doesn't work (zoom + and -)

McKlaud76, 2021-11-10:
>i have converted my dslogic to dslogicplus. now my device works with pulseview.

How can the conversion be done?

nurikk, 2021-11-11:
> > i have converted my dslogic to dslogicplus. now my device works with pulseview.
> 
> How can the conversion be done?

Check this article
https://www.eevblog.com/forum/testgear/upgrading-dslogic-basic-to-plus-without-eeprom-modification/

kpishere, 2022-02-05:
I had the same issue.  Worked a treat on 10.13 for years.  10.15, there are hiccups.  However, the advise above, IT DOES WORK.  

Select the demo device in DSView.  Then select your real devices, the light goes green and works.

Camfix, 2022-02-23:
Thank you for your answer. It worked on my computer as well.

Regards
Andreas