DreamSourceLab/DSView #344: 请问如何在win10上编译

viewtool-hhj, 2020-09-15:
我在Ubuntu中编译运行成功了，请问如何在win10系统编译呢？

hondachuang, 2021-01-13:
For most the users, we need the Windows Version. There should be official build environment to create the Windows 10 build. Can DreamSourceLab share the detail of setting up the compile environment. Thank you.

RedConda, 2021-08-31:
请问这个问题解决了吗？

hondachuang, 2021-08-31:
], 有文件可以⒖? 因槲矣 Lenovo Thinkpad Tracking point, 很y Zoom In 及 Zoom Out. 我在 Linux 上改了程式 , 加入 icons 碜 Zoom In 及 Zoom Out. 但 Windows 10 不能用

xx

RedConda, 2021-09-06:
> ��], 有文件可以�⒖�? 因�槲矣 Lenovo Thinkpad Tracking point, 很y Zoom In 及 Zoom Out. 我在 Linux 上改了程式 , 加入��� icons �碜 Zoom In 及 Zoom Out. 但 Windows 10 不能用 xx
> […](#)

你发的东西都乱码了，想请问下您解决了没有？

hondachuang, 2021-09-06:
Not yet, is there any document for reference to build the DSView on Windows?  I use Lenovo Thinkpad tracking point, it’s hard for me to zoom-in/zoom-out with tracking point. I  did a change in Linux to add icons for zoom in and zoom out, but I cannot use it in Windows.

还没, 有文件可以参考吗? 因为我用 Lenovo Thinkpad Tracking point, 很难 Zoom In 及 Zoom Out. 我在 Linux 上改了程式 , 加入两个 icons 来作 Zoom In 及 Zoom Out. 但 Windows 10 不能用 谢谢

Best Regards,
Honda

你发的东西都乱码了，想请问下您解决了没有？