DreamSourceLab/DSView #352: Windows 10 can't save .dsl file

Jumian, 2020-10-03:
[image]
windows 10 64 bit can't save .dsl file

DreamSourceLab, 2020-10-16:
@Jumian 
Could you provide us the method to reproduce this issue?
Thanks.

airbagi, 2021-01-18:
I have the same issue on many PCs on last 64-bit versions. DSLogic and DSCope. Windows 8.1, Windows 10 are installed.
When trying to save in "DSL" format this dialogue appears. Only export file types work now!

airbagi, 2021-10-15:
I've solved. The problem was in tha path using non-latin letters.
For example, if you have your username non-latin letters, there will be a problem while converting it to Unicode.
So you have to replace %APPDATA% path to the one with only latin letters.

Sure this issue must be fixed in code...