DreamSourceLab/DSView #359: Segfault running protocol decoders on Fedora 33

markfeathers, 2020-11-06:
DSView runs and gives me captures with no decoders enabled.  If I enable any decoders, it segfaults when it attempts to run:
$ DSView 
QSocketNotifier: Can only be used with threads started with QThread
Segmentation fault (core dumped)

From gdb:
Thread 23 "DSView" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff82a75640 (LWP 42496)]
0x00007ffff727cc5b in PyTuple_New () from /lib64/libpython3.9.so.1.0
Missing separate debuginfos, use: dnf debuginfo-install elfutils-libelf-0.182-1.fc33.x86_64 expat-2.2.8-3.fc33.x86_64 jasper-libs-2.0.22-1.fc33.x86_64 jbigkit-libs-2.1-19.fc33.x86_64 lcms2-2.11-2.fc33.x86_64 libdrm-2.4.102-2.fc33.x86_64 libedit-3.1-33.20191231cvs.fc33.x86_64 libjpeg-turbo-2.0.5-5.fc33.x86_64 libmng-2.0.3-12.fc33.x86_64 libtiff-4.1.0-4.fc33.x86_64 libtool-ltdl-2.4.6-36.fc33.x86_64 libwebp-1.1.0-5.fc33.x86_64 libxshmfence-1.3-7.fc33.x86_64 llvm-libs-11.0.0-1.fc33.x86_64 mesa-dri-drivers-20.2.1-2.fc33.x86_64 mesa-libEGL-20.2.1-2.fc33.x86_64 mesa-libgbm-20.2.1-2.fc33.x86_64 mesa-libglapi-20.2.1-2.fc33.x86_64 ncurses-libs-6.2-3.20200222.fc33.x86_64 python3-libs-3.9.0-1.fc33.x86_64 qt5-qtimageformats-5.15.1-1.fc33.x86_64 qt5-qtsvg-5.15.1-1.fc33.x86_64 sssd-client-2.4.0-2.fc33.x86_64
(gdb) bt
#0  0x00007ffff727cc5b in PyTuple_New () at /lib64/libpython3.9.so.1.0
#1  0x00007ffff760b041 in srd_inst_new (sess=0x7fff7c0013c0, decoder_id=0xc84870 "timing", options=0xba5c00 = {...}) at instance.c:386
#2  0x000000000067c7d4 in pv::data::decode::Decoder::create_decoder_inst(srd_session*) const (this=0x1c278e0, session=0x7fff7c0013c0) at /home/mark/Projects/DSView/DSView/pv/data/decode/decoder.cpp:152
#3  0x00000000006703dd in pv::data::DecoderStack::decode_proc() (this=0x1c240f0) at /home/mark/Projects/DSView/DSView/pv/data/decoderstack.cpp:563
#4  0x000000000067ae4d in boost::_mfi::mf0<void, pv::data::DecoderStack>::operator()(pv::data::DecoderStack*) const (this=0xe51a68, p=0x1c240f0) at /usr/include/boost/bind/mem_fn_template.hpp:49
#5  0x000000000067adb1 in boost::_bi::list1<boost::_bi::value<pv::data::DecoderStack*> >::operator()<boost::_mfi::mf0<void, pv::data::DecoderStack>, boost::_bi::list0>(boost::_bi::type<void>, boost::_mfi::mf0<void, pv::data::DecoderStack>&, boost::_bi::list0&, int) (this=0xe51a78, f=..., a=...)
    at /usr/include/boost/bind/bind.hpp:259
#6  0x000000000067ad61 in boost::_bi::bind_t<void, boost::_mfi::mf0<void, pv::data::DecoderStack>, boost::_bi::list1<boost::_bi::value<pv::data::DecoderStack*> > >::operator()() (this=0xe51a68) at /usr/include/boost/bind/bind.hpp:1294
#7  0x000000000067acd0 in boost::detail::thread_data<boost::_bi::bind_t<void, boost::_mfi::mf0<void, pv::data::DecoderStack>, boost::_bi::list1<boost::_bi::value<pv::data::DecoderStack*> > > >::run() (this=0xe51930) at /usr/include/boost/thread/detail/thread.hpp:120
#8  0x00007ffff7f64fa8 in boost::(anonymous namespace)::thread_proxy(void*) (param=<optimized out>) at libs/thread/src/pthread/thread.cpp:179
#9  0x00007ffff7f393f9 in start_thread () at /lib64/libpthread.so.0
#10 0x00007ffff6284b03 in clone () at /lib64/libc.so.6

So the segfault is here:
https://github.com/DreamSourceLab/DSView/blob/13ace4b4117ad498484c10f961affec3e95aafac/libsigrokdecode4DSL/instance.c#L386

(gdb) print *di
$3 = {decoder = 0xa02400, sess = 0x7fff7c001a20, py_inst = 0x0, py_pinvalues = 0x0, inst_id = 0x7fff7c001920 "timing-1", pd_output = 0x0, dec_num_channels = 1, dec_channelmap = 0x7fff7c000c50, next_di = 0x0, condition_list = 0x0, match_array = 0, abs_start_samplenum = 0, abs_end_samplenum = 0, 
  inbuf = 0x0, inbuf_const = 0x0, inbuflen = 0, abs_cur_samplenum = 0, abs_cur_matched = 0, old_pins_array = 0x0, thread_handle = 0x0, got_new_samples = 0, handled_all_samples = 0, want_wait_terminate = 0, first_pos = 0, skip_zero = 0, decoder_state = 0, got_new_samples_cond = {p = 0x0, i = {0, 
      0}}, handled_all_samples_cond = {p = 0x0, i = {0, 0}}, data_mutex = {p = 0x0, i = {0, 0}}}

Let me know if I can provide any more information.

imi415, 2020-12-06:
Can confirm the same issue occurs on Arch Linux (Python 3.9) when decoders are used.

Electro707, 2020-12-08:
Can confirm was well with @imi415 with decoders, even with the stable release. I have gdb debugged, and traced the fault to the function srd_inst_new in instance.c in the decoder program.

Electro707, 2020-12-08:
I fixed it. I simply moved ```gstate = PyGILState_Ensure();``` from line 392 to line 372 (Makes sense, as ```PyTuple_New``` should be called when the thread can call Python functions)

stecman, 2020-12-12:
Can confirm @Electro707's patch fixes this crash (Python 3.9, Debian 11).

My crash stacktrace was slightly different, but that change fixed it:

```
#0  0x00007fe0cdc81223 in PyTuple_New () at /lib/x86_64-linux-gnu/libpython3.9.so.1.0
#1  0x00007fe0ce222031 in srd_inst_newPython Exception <class 'gdb.error'> There is no member named keys.: 
 (sess=sess@entry=0x7fe09841a9d0, decoder_id=0x5629db650390 "0:uart", options=options@entry=0x5629db92d1e0) at instance.c:386
#2  0x00005629da0831dd in pv::data::decode::Decoder::create_decoder_inst(srd_session*) const (this=0x5629dc38fe80, session=0x7fe09841a9d0) at /home/stecman/sources/DSView/DSView/pv/data/decode/decoder.cpp:152
#3  0x00005629da07f09e in pv::data::DecoderStack::decode_proc() (this=0x5629dc390600) at /usr/include/boost/smart_ptr/shared_ptr.hpp:732
#4  0x00007fe0ceb659b7 in  () at /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#5  0x00007fe0ceb3dea7 in start_thread (arg=<optimized out>) at pthread_create.c:477
#6  0x00007fe0ccb64d4f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

buha, 2021-01-19:
I confirm this fix works for me too. Please merge PR.

PR #366 (2021-01-22): Segmentation Fault Fix
This PR fixed #359 by moving ```gstate = PyGILState_Ensure();``` to before ```PyTuple_New``` is called, getting rid of the segmentation fault causing by the uncertainty of the GIL when calling ```PyTuple_New```.
closed 2021-01-22 by DreamSourceLab