DreamSourceLab/DSView #386: SPI 协议配置为 CPOL=1,CPHA=1 时解码存在问题

anydream, 2021-03-09:
[image]

正确的解码方式应该是从 SCK **第二个**上升沿开始计算, 而不是第一个

DreamSourceLab, 2021-03-11:
SPI的4种模式如下图所示：
[image]
截图里采集到的信号，实际上来讲不符合任何一种模式。
如果是CPOL=1，CPHA=1。clk空闲时应为高电平，也就是cs下降沿的时候clk应该为高。

anydream, 2021-03-15:
这个是STM32的SPI时序图, 实测GD32的也是这样, SCLK不是立即拉高的

DreamSourceLab, 2021-03-19:
我们只能根据协议的官方specification来解码，而不是某一个芯片的某一次输出。
比如：https://www.analog.com/en/analog-dialogue/articles/introduction-to-spi-interface.html
实际上ST公司自己的文档介绍SPI也是上图的时序。
比如：https://www.st.com/content/ccc/resource/training/technical/product_training/group0/3e/ee/cd/b7/84/4b/45/ee/STM32F7_Peripheral_SPI/files/STM32F7_Peripheral_SPI.pdf/_jcr_content/translations/en.STM32F7_Peripheral_SPI.pdf
当然如果只是解决你现在遇到的问题，只需要设置一个解码起始位置即可。具体可以参考我们手册关于区域解码的内容。
如果再进一步的话，我猜测你发的时序图只是一个软件模拟SPI的行为，而不是硬件NS和模式设置的自动输出。如果我猜测错误的话，你应该向ST公司求助，需要他们解答为什么实际波形和官方文档的描述不一致。

closed 2021-03-23 by anydream