DreamSourceLab/DSView #397: Ubuntu 21.04 drops QT-4 support, unable to make. Do you plan to upgrade?

priv, 2021-04-23:
build log as followed:
```
Make Warning (dev) at /usr/share/cmake-3.18/Modules/GNUInstallDirs.cmake:225 (message):
  Unable to determine default CMAKE_INSTALL_LIBDIR directory because no
  target architecture is known.  Please enable at least one language before
  including GNUInstallDirs.
Call Stack (most recent call first):
  CMakeLists.txt:25 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /usr/share/cmake-3.18/Modules/FindQt4.cmake:1314 (message):
  Found unsuitable Qt version "" from NOTFOUND, this code requires Qt 4.x
Call Stack (most recent call first):
  CMakeLists.txt:90 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/alvin/Downloads/DSView-1.12/DSView/CMakeFiles/CMakeOutput.log".
```

prosper00, 2021-06-15:
I can't get it to build on 21.04 either. Seems to have issues with qt5.15. I get as far as 'make' which dies shortly after this error:

    /home/bdroy/Downloads/DSView-1.12/DSView/pv/view/viewport.cpp:1555:38:   ERROR : variable ‘QPainterPath path’ has initializer but incomplete type
    1555 |             QPainterPath path(QPoint(_cur_preX, _cur_preY));

prosper00, 2021-06-15:
Update - well, I couldn't get it to build using the zip from the main website. But if I git clone this repo, it builds and installs just fine (with a bunch of 'deprecated' warnings here and there).

meeas, 2021-08-21:
Confirmed.  The downloadable zip fails to build on the latest Debian and Ubuntu distributions, however if you pull the latest git repo, it builds fine.  

And as the initial post remarked, qt5-default  is no longer available on Debian and Ubuntu.  Instead, use the following command to install the needed dependencies:

`sudo aptitude install git-core build-essential cmake autoconf automake libtool pkg-config libglib2.0-dev libzip-dev libudev-dev libusb-1.0-0-dev python3-dev libboost-dev libboost-test-dev libboost-thread-dev libboost-system-dev libboost-filesystem-dev check libfftw3-dev qtbase5-dev qttools5-dev`

May I suggest to DS to consider creating an AppImage for this package people can download a standalone file and just run it regardless of the distribution they are on?  This is what Saleae is doing with their new Logic2 app, and it seems to work great!

priv, 2022-11-07:
INSTALLL has been updated with qtbase5-dev information

closed 2022-11-07 by priv