DreamSourceLab/DSView #398: The SPI flash decoder can't handle long SPI transfers.

Crest, 2021-04-23:
I'm using DSView to capture the SPI traffic between a STM32H750 and a SPI flash chip (Winbond W25Q*). To improve transfer efficiency my driver uses DMA transfers of up to 64kiB. The SPI flash protocol decoder isn't written with such long transfers in mind and slows down to such a degree the DSView becomes unusable. The symptoms manifest like this:

- the user interface takes >20 seconds to respond to scrolling
- 100% on one core for several minutes after capturing less than 100ms on four pins

Changing the SPI flash driver to read the same data in proportionally more smaller transfers from the chip decodes as expected.

DreamSourceLab, 2021-04-25:
@Crest 
Could you help to upload the dsl file?
Thanks.

ma6254, 2021-08-17:
[DSLogic-la-210817-212916.zip](https://github.com/DreamSourceLab/DSView/files/7000744/DSLogic-la-210817-212916.zip)

Something similar happened to me
I grabbed the spi data of w25q16jv 2Mbyte, clk about 10Mhz, 2Mbyte, and then decoded it. It took a few minutes for dsview to complete the analysis, but it couldn’t be dragged to view.

也是一样的情况，器件是w25q16jv，时钟约为10Mhz，读取了全部的2Mbyte数据
启动解析之后，花了好几分钟才100%，但是软件很卡，没办法拖动看波形，试了几次有概率会闪退或无响应
环境：
- 笔记本电脑：联想Y7000，I7-9750H, RAM 40G，GTX1650 4G
- 系统：win10
- DSView：V1.1.2

[image]
[image]