DreamSourceLab/DSView #399: Linear RMS range for FFT (feature request)

fgl27, 2021-05-06:
In the FFT Math option, using linear RMS for the Y-axis Mode, is a great feature but the horizontal range is too big for most signal, making it hard to see the levels, the odd thing is that it gets bigger if you change the volt scale of the channel.

See here how small it's.

[image]

Can you make it a option to change the range so is more visible in the Y-axis?

Thanks.

DreamSourceLab, 2021-05-07:
@fgl27 
For the horizontal range, you can use Mouse wheel for zoom in/out.
The range will be half of current sampling rate.
For the vertical range, it depends on the ratio of the waveform amplitude to the wave window.

fgl27, 2021-05-08:
I understand all of that.

But I wanna to see the "whole picture", both of my signal and a FFT on the Linear RMS mode that fully fill up the Y-axis, in order to me to notice on that case all the frequencies on the FFT without any zoom.

But for that is needed a way to change the Y-axis range, just like you already can do to the DVB mode.

I assume is possible to have a simple multiplier there, replacing/complementing the DVB range as a "Y-range" that has different values for it mode .

Thanks,