DreamSourceLab/DSView #404: DSView v1.12 for MacOS capture failed.

mgyuli, 2021-06-10:
在软件打开的情况下， 连接usb后自动切换到所连接设备。但切换过程中并未初始化U2P20的参数，设备持续显示红灯，从而导致capture failed。 手动选择Demo Device后再切换到U2P20后，设备估计完成初始化显示绿灯， 然后功能一切正常。

ikey-ru, 2021-09-28:
Same problem. I downloaded a new version of the application and cannot work. 
First: Please replug device to refresh hardware configuration!
After reboot device:  Failed to start session. Ver.1.0.1 - working.

mbrossard, 2021-12-30:
I had a similar issue with DSLogic Plus. I fixed it by downloading `DSLogicPlus.fw` from https://github.com/DreamSourceLab/DSView/tree/886b847c21c606df3138ce7ad8f8e8c363ee758b/DSView/res
and copying it in `/Applications/DSView.app/Contents/MacOS/res/`

ikey-ru, 2021-12-30:
It really worked! Thank you.

ikey-ru, 2021-12-30:
This worked until the first disconnection of the device from USB. After reconnecting, the problem returned (

ikey-ru, 2022-01-15:
Unfortunately this is very unstable. Apparently, it only works until the device is turned off. After rebooting iMac or turning off/on the device again the same problem.