DreamSourceLab/DSView #406: ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper'

benner, 2021-06-14:
I've build DSView from source (last commit: 2852876e988c8f8764406ff5789bb9d493c095f7)

When starting it I see:

```
% DSView 
srd: Failed to load decoder sle44xx: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

srd: Failed to load decoder ir_sirc: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

srd: Failed to load decoder sle44xx: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

srd: Failed to load decoder ir_sirc: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

Protocol decoder enabled!

Starting a hotplug thread...
```

OS: Ubuntu 20.10
Python: 3.8.6

mhgue, 2021-07-27:
Same issue with Ubuntu 20.04.2 LTS and Python 3.8.10 using self compiled DSView 1.1.2

mhgue, 2021-07-27:
Just replaced bitpack_lsb( XYZ,0) by bitpack(XYZ):
```
diff --git a/libsigrokdecode4DSL/decoders/ir_sirc/pd.py b/libsigrokdecode4DSL/decoders/ir_sirc/pd.py
index 14ba63f..02ff2b5 100644
--- a/libsigrokdecode4DSL/decoders/ir_sirc/pd.py
+++ b/libsigrokdecode4DSL/decoders/ir_sirc/pd.py
@@ -17,7 +17,7 @@
 ## along with this program; if not, see <http://www.gnu.org/licenses/>.
 ##
 
-from common.srdhelper import bitpack_lsb
+from common.srdhelper import bitpack
 from .lists import ADDRESSES
 import sigrokdecode as srd
 
@@ -168,8 +168,8 @@ class Decoder(srd.Decoder):
                     raise SIRCError('incorrect bits count {}'.format(len(bits)))
                 break
 
-        command_num = bitpack_lsb(command, 0)
-        address_num = bitpack_lsb(address, 0)
+        command_num = bitpack(command)
+        address_num = bitpack(address)
         command_str = '0x{:02X}'.format(command_num)
         address_str = '0x{:02X}'.format(address_num)
         self.putg(command[0][1], command[-1][2], Ann.CMD, [
@@ -182,7 +182,7 @@ class Decoder(srd.Decoder):
         ])
         extended_num = None
         if extended:
-            extended_num = bitpack_lsb(extended, 0)
+            extended_num = bitpack(extended)
             extended_str = '0x{:02X}'.format(extended_num)
             self.putg(extended[0][1], extended[-1][2], Ann.EXT, [
                 'Extended: {}'.format(extended_str),
diff --git a/libsigrokdecode4DSL/decoders/sle44xx/pd.py b/libsigrokdecode4DSL/decoders/sle44xx/pd.py
index 9f33207..75d85dc 100644
--- a/libsigrokdecode4DSL/decoders/sle44xx/pd.py
+++ b/libsigrokdecode4DSL/decoders/sle44xx/pd.py
@@ -17,7 +17,7 @@
 ## along with this program; if not, see <http://www.gnu.org/licenses/>.
 ##
 
-from common.srdhelper import bitpack_lsb
+from common.srdhelper import bitpack
 import sigrokdecode as srd
 
 class Pin:
@@ -431,7 +431,7 @@ class Decoder(srd.Decoder):
         # a field? To tell CTRL/ADDR/DATA of a CMD entry apart?
         bits = self.bits
         self.bits = []
-        data = bitpack_lsb(bits, 0)
+        data = bitpack(bits)
         ss = bits[0][1]
         es = bits[-1][2]
```
Not sure if correct but starting without errors.

ardnew, 2021-08-15:
To resolve this issue, I just merged those two functions (found in upstream `libsigrokdecode`). This is probably the correct solution, as there is no logic that needs to be modified anywhere:

```diff
index 6c45af9..b56cce6 100755
--- i/libsigrokdecode4DSL/decoders/common/srdhelper/mod.py
+++ w/libsigrokdecode4DSL/decoders/common/srdhelper/mod.py
@@ -30,8 +30,22 @@ def bin2int(s: str):

 def bitpack(bits):
     return sum([b << i for i, b in enumerate(bits)])

+def bitpack_lsb(bits, idx=None):
+    '''Conversion from LSB first bit sequence to integer.'''
+    if idx is not None:
+        bits = [b[idx] for b in bits]
+    return bitpack(bits)
+
+def bitpack_msb(bits, idx=None):
+    '''Conversion from MSB first bit sequence to integer.'''
+    bits = bits[:]
+    if idx is not None:
+        bits = [b[idx] for b in bits]
+    bits.reverse()
+    return bitpack(bits)
+
 def bitunpack(num, minbits=0):
     res = []
     while num or minbits > 0:
         res.append(num & 1)
```

This will probably resolve itself after this branch merges upstream changes
> _**Lookin' at you, devs…**_  :wink:

tedy58, 2021-11-20:
The same issue with Ubuntu 20.04.3 LTS