DreamSourceLab/DSView #408: Enhancement Request: User-defined labels on channels and User-defined colours on channels

egrabczewski, 2021-06-18:
[image]

ENHANCEMENT REQUEST:
1. Could be have the option to type-in user-defined labels on all channels
2. Could the user-defined colurs also change the background colour of the channel number (as shown on the uploaded image)

DreamSourceLab, 2021-06-18:
@egrabczewski 
Just click the channel name, then you can type-in user-defined labels on all channels.
Pelase refer to section: 2.6.4 Channel Customization of our user guide:
[User Guide](https://www.dreamsourcelab.com/doc/DSView_User_Guide.pdf)

egrabczewski, 2021-06-18:
Yes, on page 24 it says how to assign a name. Thanks for that. 
However, the background colour of the channel number does not change to the user-defined Colour. 
So could we have that enhancement please.