DreamSourceLab/DSView #422: CSV Precision too low

stevenlovegrove, 2021-08-16:
I'm using the data acquisition feature to capture a signal in the mV range. The signal is captured correctly in DSView, but when I export using the CSV feature to get the data into a simulation tool the precision is much too low. 

Attached is an example with the saved DSL file from a short capture and the exported CSV file (with just 2 decimal places)
[Archive.zip](https://github.com/DreamSourceLab/DSView/files/6989118/Archive.zip)

With this export, it is not possible to run simulations on the captured signal.

Thanks!

stevenlovegrove, 2021-08-17:
Maybe this is actually fixed already here:
https://github.com/DreamSourceLab/DSView/commit/26db5b2cb2de0c39e462ee36c9100e83b2560bcc#diff-418e9a81993e4b601b0f3178b9d81bc0d5a6db47bed38ea9027289cfb957ba61R267 ?

Is there a daily build anywhere to save me having to compile this?

DreamSourceLab, 2021-08-18:
@stevenlovegrove 
Yes, it's fixed.
Sorry, we don't have a daily build.

stevenlovegrove, 2021-08-18:
Thanks for your response, @DreamSourceLab .

Would you consider taking a Pull Request if I were to get around to setting up a Github Action to build for MacOS?

closed 2021-08-18 by stevenlovegrove