DreamSourceLab/DSView #426: DSView builds and runs on FreeBSD 13.0

Crest, 2021-08-31:
I tried to builds DSView on FreeBSD 13 and was pleasantly surprised that it did it almost compiled out of the box. It got to `cd DSView && cmake .` and required the workaround recommended for Fedora to find `libsigrok4DSL` and `libsigrokdecode4DSL` under the `/usr/local` prefix. The next error was that the libusb version included in the FreeBSD base system is too old. It ships with version 1.0.13 instead of 1.0.16. Reducing the required version was enough to build and install DSView. It runs and detects my DSLogic U3Pro16. I haven't gotten around to testing how well it works.

Crest, 2021-08-31:
But it hangs at "100%" after capturing just a few ms at 1MHz.

ivan-volnov, 2025-02-28:
Did you fix your issue?