DreamSourceLab/DSView #428: Use upstream libsigrokdecode

fornellas, 2021-09-01:
I'm not sure the circunstances behind having forked [libsigrokdecode](https://sigrok.org/wiki/Libsigrokdecode) into `libsigrokdecode4DSL`, but it is causing a lot of headaches:

- Original protocol decoders that "just worked" such as UART were substituted by 2 replacements, one which [is broken](https://github.com/DreamSourceLab/DSView/pull/427) and the other that removes the ability to see RX from TX (why?).
- Codebase here lags behind upstream and [peolpe are trying to backport stuff](https://github.com/DreamSourceLab/DSView/pull/421) as the fork lags behind and breaks things.
- Protocol decoders [added here](https://github.com/DreamSourceLab/DSView/pull/338) don't make upstream.
- Protocol decoders added upstream don't make it here.
- The interface of [`self.matched`](https://sigrok.org/wiki/Protocol_decoder_API/Queries#self.matched) is broken:
  - It is expected to return a tuple of booleans, but it is returing an integer.
  - While some [protocol decoders](https://github.com/DreamSourceLab/DSView/blob/master/libsigrokdecode4DSL/decoders/1-i2c/pd.py#L267) make use of it this way, various others [don't](https://github.com/DreamSourceLab/DSView/blob/master/libsigrokdecode4DSL/decoders/ieee488/pd.py#L627), so they blow up with `TypeError: 'int' object is not subscriptable`.
    - This includes any upstream protocol decoders from working here.

I'm after using a simple UART decoder, to develop another decoder on top. What should be simple, is becoming a huge pain.

Can't we have `libsigrokdecode` being used here? It can work fine as a [Git submodule](https://git-scm.com/book/en/v2/Git-Tools-Submodules).

abougouffa, 2021-09-17:
Totally agree!

fornellas, 2021-09-27:
I did an "proof of concept" on how hard it'd be to make use of ustream `libsigrokdecode` it turns out it is somewhat simple: https://github.com/fornellas/DSView/tree/drop_libsigrokdecode4DSL.

It shouldn't be too hard fixing the problems below, though I'm not sure if there will be willingness to merge such PR by @DreamSourceLab, so I'm not gonna wrap this up for now.

[image]

DreamSourceLab, 2022-09-05:
@fornellas 
This is an important performance improvement, not destruction. 
Take the following data file as an example. 
[uart.zip](https://github.com/DreamSourceLab/DSView/files/9488955/uart.zip)
On the same computer, it takes more than 620 seconds to parse this data file using pulseview's UART decoder, which occupies more than 3 gigabytes of memory.
For the same data, using DSView's 0: UART decoder, the parsing time is only 120 seconds, which is only 1 / 5 of the original, and the memory occupied is only more than 300 m, which is only 1 / 10 of the original. 
There is no reason for us to abandon this better decoder implementation.

fornellas, 2022-09-05:
@DreamSourceLab I believe your previous comment was miwed up with discussions from #427 and  #429. Let's keep those discussions there.

This issue here is regarding the fact that DSView forked & patched Sigrok libraries (libsigrokdecode), in a way that's incompatible with Sigrok decoders. These changes created a phethora of issues, which made my life a LOT harder when attempting to develop for DSView.

As I poked around the code, it seems the changes were minimal, and potentially, something reversible, so we can use standard libsigrokdecode (and all current & future decoders there).

DreamSourceLab, 2022-09-06:
@fornellas 
uart.sr file is prepared for pulseview, so you can test the result.
The following file is dsl with same data.
[uart_dsl.zip](https://github.com/DreamSourceLab/DSView/files/9492733/uart_dsl.zip)

We have never declared any compatibility, the open source license allows any modification, not to mention the improvement of performance and user experience
We just want to give customers a better experience, not to struggle with compatibility with some APIs.

fornellas, 2022-09-07:
> We just want to give customers a better experience, not to struggle with compatibility with some APIs.

Unfortunately my experience was exactly the opposite: the incompatible API changes pushed, while 100% within what the license allows, created a much worse user experience for me. As DSView does not provide full in depth documentation for its decoder API interface, and I had to rely on Sigrok's, which caused all the confusion, due to the breaking changes. Also, there were bugs at DSView codebase resulting from API changes, that I followed as examples to my code, and were actually broken. It seem these [have been fixed](https://github.com/DreamSourceLab/DSView/commit/5b1cbbb9e9db9d0de1aa96dec80d859d50aeb966) (`self.matched` changes).

By the end of the day, you can do whatever you prefer with DSView decoder API. My suggestion (and from various other users who thumbed up here) is to keep the API compatible with Sigrok. You can take it or leave it.