DreamSourceLab/DSView #430: Fix broken self.matched interface

fornellas, 2021-09-01:
As [documented upstream](https://sigrok.org/wiki/Protocol_decoder_API/Queries#self.matched), the interface of `self.matched` for a decoder must return a tuple of booleans. It is currently returningn an integer.

While some [protocol decoders](https://github.com/DreamSourceLab/DSView/blob/master/libsigrokdecode4DSL/decoders/1-i2c/pd.py#L267) make use of it this way, various others [don't](https://github.com/DreamSourceLab/DSView/blob/master/libsigrokdecode4DSL/decoders/ieee488/pd.py#L627), so they blow up with `TypeError: 'int' object is not subscriptable`.

This is also true for [all upstream decoders](https://sigrok.org/gitweb/?p=libsigrokdecode.git;a=tree;f=decoders;hb=HEAD), which can't work here, unless they are patched (eg: https://github.com/DreamSourceLab/DSView/pull/429)

fornellas, 2022-09-05:
@DreamSourceLab while I acknowledge this may have been originally thought to have been an improvement, I don't believe this is the case now.

This issue's description gives clear evidence that this incompatible API change in relation to Sigrok's API broke existing decoders **at DSView's codebase**. Also, it makes any stacked protocol decoders from Sigrok unusable at DSView, taking away any hopes of collaboration.

Would you please review the description, perhaps I can help with more details if required.

DreamSourceLab, 2022-09-06:
@fornellas 
we don't want to use the original interface, because it has speed and performance problems.
And I don't know why you said any stacked protocol decoders from Sigrok unusable at DSView.
There are dozens of stacked decoders in DSView, which can work well.

fornellas, 2022-09-07:
> And I don't know why you said any stacked protocol decoders from Sigrok unusable at DSView.

@DreamSourceLab I think I got part of the confusion here. This issue was cut a long time ago, and back then,  [this change](https://github.com/DreamSourceLab/DSView/commit/5b1cbbb9e9db9d0de1aa96dec80d859d50aeb966), that fixes part of the problem referred at the description, was not there, so I see why it was confusing now "this is not actually broken". So, things WERE broken, things WERE fixed, and now, the matter is an API change (below).

---

So, moving on, it seems we have 2 things on the table:

1. There were changes pushed to improve speed (fair).
2. The current interface of `self.matched` returns an `int`, and this diverges from [Sigrok's API](https://sigrok.org/wiki/Protocol_decoder_API/Queries#self.matched), where it is a tuple of booleans.

It is not clear to me if (2) is a requirement for any performance gains here or not. Whatever the case, the fact is that, DSView protocol decoder API is hyper similar to Sigrok's, but NOT exactly the same, due to the change at `self.matched`. And this change, created me a lot of headache in the past, as it is not super clear that it exists.

Do you think it would be possible to keep the speed benefits pushed, while having `self.matched` be the same tuple of booleans as Sigrok's API? Doing so would be great, in the sense that DSView would have exactly the same API as Sigrok's, and protocol decoders could be painlessly shared between both.

WDYT?