DreamSourceLab/DSView #435: DSView 1.12, Trigger position does not work.

m72m72, 2021-10-03:
Hello!

DSView 1.12, Windows 7x64, u3pro32

Sometimes it works fine. But in most cases the value in the digital field does not correspond to the slider. Also position in the capture is very strange.

With best regards, Mikhail.
[image]

skiphansen, 2022-07-18:
I have a similar issue here with version 1.2.0 built from today's (July 18, 2022) git.  I can set the trigger to about 30% but that's it.  I really need 99% to capture the issue I'm trying to debug.

DreamSourceLab, 2022-07-20:
@m72m72 
The trigger position will be at the setting position only when trigger event happend after setting position.
For example, if the first sample hit the trigger, the trigger position must be 0, because there are no any samples before the first sample.

m72m72, 2022-08-22:
Hello!

Version 1.2.0. Windows 7 x64.

Trigger position still not works properly with 1G samples. With 100...500M samples works well.
Please trace the way of this variable (overload and etc) in program and his limits in the GUI.
The slider and input field works strange. Sometimes I can not enter value more than 13%. :-(
After five retries (turn on/turn of/restart software) I have set trigger position to 90%. But only 14% or 36% of record was captured before trigger. There is only two values 14% and 36% it is also strange.

I have a test signal from Digilent Digital Discovery. Three lines:
0. Delay 2.0s, clock 1 kHz
1. Delay 1.9s, clock 2 kHz
2. Delay 1.8s, clock 5 kHz
Trigger is "RISING #0"

So there is no trigger can occurs before 90% of record.

With best regards, Mikhail.

m72m72, 2022-08-22:
[skiphansen](https://github.com/skiphansen)
> I have a similar issue here with version 1.2.0 built
I have just tested in different modes. So the trigger position 90% works well only when record length is lower than "const==256M" samples. It is very simple to overflow with 2Gb buffers in U3Pro16/U3Pro32 and 1..7 active channels. RLE does not matter.

closed 2022-08-23 by m72m72