DreamSourceLab/DSView #441: How to compile for windows

maruncz, 2021-11-12:
hi, i would like to run modified DSView on windows, but i struggle with compilation.
i tried script from sigrok-utils, but it doesn't work, it cannot find some libraries even if i installed them
does anybody know how to compile it?

RedConda, 2021-12-23:
Have you solved this problem?

maruncz, 2021-12-23:
no

RedConda, 2021-12-23:
This problem is quite challenging, I have been stuck for a long time, so far not solved

christianeisendle, 2022-01-04:
Hi @maruncz,

i'm struggling with the same issue. I tried to compile using the cross compilation scripts in sigrok-utils (which works perfectly fine for pulseview). I had to tweak it a bit because the Cmake script doesnt work out of the box:

- project(DSView) in CMakeLists.txt needs to come first, at least before include(FindPkgConfig) on line 24
- for some reason pkg-conf failed to find libsigrok4DSL and libsigrokdecode4DSL (which compiled fine on my end and also .pc files were created). I had to fix it by defining PKG_CONFIG_EXECUTABLE variable in CMakeLists.txt and let it point to the x-compiled version of pkg-config (strangely enough, this is not needed when building pulseview)
- DSView.rc (the windows icon) is missing in the repository even though CMakeLists.txt refers to it

So finally it started to compile but it then fails in main.cpp where it uses some LPWSTR types which are a MS c compiler specific thing. So in other words: I feel that compilation for windows only works natively under windows with Visual Studio installed. Sadly enough, there is no documentation or recipe available on how to do that. Would be nice if the guys from @DreamSourceLab could update the Readme to add instructions on how to compile for windows.

andrei7c4, 2024-01-09:
I have the same question as others in this thread. Could @dreamsource-tai please provide at least some kind of guide for building Windows executable? Native or cross-compile, doesn't matter to me.

maxdd, 2026-01-31:
Any update guys? I also need to build a dedicated version