DreamSourceLab/DSView #447: v1.12 decodes SPI erroneously

mehmetalianil, 2021-12-01:
It seems to me that SPI mode 3 (CPOL 1 CPHA 0) is decoded with one bit errors. 
Here in this image, MISO 0x80 should be 0xA0 as seen in the waveforms. I always get slightly off errors, and with each run the places of these errors change. 

I am using 1.12 and using the Zadig WinUSB driver.

[image]
[image]

mehmetalianil, 2021-12-01:
I tried v1.1.1, and it decodes the right value.
(There seems to be a glitch issue on 1.1.2, too. I don't have that here although I did not touch the setup, really.)
[image]

mehmetalianil, 2021-12-01:
v1.1.1 has also an issue. This should have been a  0x58 0x80
[image]

DreamSourceLab, 2021-12-02:
@mehmetalianil 
This spi wave is in mode (CPOL 1 CPHA 1), not (CPOL 1 CPHA 0)
Just change to the right mode, you will get the right decode results.
Attached is the SPI mode introduction.
[image]

mehmetalianil, 2021-12-02:
I always set it as CPOL 1 CPHA 0 (but good to check) and it is can be seen that other bytes are interpreted correctly with respect to MODE 3.