DreamSourceLab/DSView #459: cant save file

kzh92, 2021-12-23:
hello,
I have cleanly installed DSView.
This is the version of my DSView.
[image]

After capturing, I can't save my data.
This is the error message.
[image]

How can I solve this problem?
Thanks in advance.

christianeisendle, 2022-01-04:
This issue was already reported multiple times and the guys from @DreamSourceLab have no idea how to fix it.
I guess it is related to the fact the DSView stores the data/traces into a zip archive but the way how it's done is not super efficient: Long traces can easily get several gigabyte of data (because they are stored as raw samples in the zip), but the samples are added to the zip file in chunks of 2MB. In other words, if you have a trace of 2GB (which would be equivalent of sampling e.g. 10sec. at 100MHz) 1000 chunks of 2MB are added to the zip file. The zip file is opened and closed for every chunk added and i guess this might cause the issue.

I have some ideas on how to fix it, but i see currently no way to get DSView compileable under windows (see  issue #441 )

kzh92, 2022-01-04:
thank you for your kind answer, @christianeisendle.

closed 2022-01-04 by kzh92