DreamSourceLab/DSView #468: Debian 11 isssue DSView doesn't install

tedy58, 2022-01-07:
I am newbie in git.
I try to compile it on fresh Debian 11.2.0 (bullseye) according to INSTALL instructions in the main directory. Installation of libraries and dependencies and the main two steps goes well, but when try to cmake in the ~/DSView/DSview i have an error. 
According to bug reports in Debian  wiki (https://wiki.debian.org/Qt4Removal) it has been removed since 2020-03-07. 

~/DSView/DSView$ cmake .
CMake Error at /usr/share/cmake-3.18/Modules/FindQt4.cmake:1314 (message):
  Found unsuitable Qt version "5.15.2" from /usr/bin/qmake, this code
  requires Qt 4.x
Call Stack (most recent call first):
  CMakeLists.txt:90 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/user/DSView/DSView/CMakeFiles/CMakeOutput.log".

I appreciate any help on that issue.

liuhuihz, 2022-01-18:
The Qt5 development packages weren't be found, so fallback to Qt4.
Install the related Qt5 development packages, like qtbase5-dev and so on.
Check this link:
https://askubuntu.com/questions/508503/whats-the-development-package-for-qt5-in-14-04

tedy58, 2022-06-18:
I have install now a new fresh Debian 11 and fulfill the dependencies according to the Requirements in the main INSTALL instructions. I download the fresh code from git. Then compile with `cmake .` and get lot of warnings, but finally it compile. Now I see that it is DSView 1.2.0-RC8.
So it does not have any more "Stack decode", as it was in 1.1.2 release. How can be added a new decoder. for example:
  'winbond': {
      0x17: 'W25Q128BV',
    },
[spiflash.tar.gz](https://github.com/DreamSourceLab/DSView/files/8933276/spiflash.tar.gz)
I see now that it is add as protocol, but no commands when decode the stream or buffer.