DreamSourceLab/DSView #472: CAN remote request - decoder error

hurisson, 2022-02-07:
Hello! I noticed such an error in the CAN decoder - with a remote request (RTR = 1) and DLC = 2, I see an incorrect decryption, the CRC field and all subsequent signs should go after the DLC. In the diagram I see DB0, DB1 - contradicts the CAN standard when RTR is requested

[image]

hurisson, 2022-02-07:
### For example:
[image]

I attach an archive with temporary DSL diagrams
[Ds_example.zip](https://github.com/DreamSourceLab/DSView/files/8013904/Ds_example.zip)

JacobCrabill, 2022-03-10:
I'm having the same issue for all CAN messaging.  The decoder doesn't actually wait for the start of the frame, and instead starts decoding from whenever the recording happened to start (it'll even decode a dead bus with no traffic as continuous frames of all 0).

MattWoodhead, 2023-01-01:
> I'm having the same issue for all CAN messaging. The decoder doesn't actually wait for the start of the frame, and instead starts decoding from whenever the recording happened to start (it'll even decode a dead bus with no traffic as continuous frames of all 0).

I have realised after some experimentation that this happens if you connect to CAN High, if you connect to CAN Low, it will trigger correctly. Pulseview behaves exactly the same.