DreamSourceLab/DSView #475: 解码协议中无法选择UART

bowenzhangj, 2022-03-04:
环境：Windows 10 专业版
安装包：DSView_v1.1.2_x64_setup.exe
卸载应用重装，还是无法使用UART解码
（几个月前还是可以使用UART解码的，现在突然就不行了）
[image]

dreamsource-tai, 2022-03-07:
@bowenzhangj
请问，你说的UART不能用，它的具体表现是什么样的问题呢？

dreamsource-tai, 2022-03-07:
@bowenzhangj 
我看到了你文件夹的截图，说明UART协议是存在的。但是在的协议下拉列表中是看不到的。从之前能正常使用到现在不能使用，这中间你有做过什么吗？有没有修改协议的代码？

bowenzhangj, 2022-03-07:
> @bowenzhangj 我看到了你文件夹的截图，说明UART协议是存在的。但是在的协议下拉列表中是看不到的。从之前能正常使用到现在不能使用，这中间你有做过什么吗？有没有修改协议的代码？

没有修改代码，安装应用之后就没有手动去更改安装文件夹里面的东西。之前装了多个Python版本有出现版本冲突，就卸载重装Python了；还有我前段时间安装过Visual Studio 2022，安装和更新了一些组件；还有一个可能是公司电脑加装了加密系统，会对文件进行加密，或者限制电脑的一下功能权限，不知道是不是这影响的

dreamsource-tai, 2022-03-07:
@bowenzhangj
请加我的微信: tzz8080 ，我们进一步沟通。我有一个大的附件想发到你的邮箱，但是发不了。

bowenzhangj, 2022-03-07:
日志：
Error processing line 7 of C:\Python\Python37\lib\site-packages\pywin32.pth:

  Traceback (most recent call last):
    File "site.py", line 168, in addpackage
    File "<string>", line 1, in <module>
  ValueError: source code string cannot contain null bytes

Remainder of file ignored
srd: Failed to load decoder 0-uart: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\0-uart\__init__.py", line 40, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\0-uart\pd.py", line 22, in <module>
    from common.srdhelper import bitpack
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder 1-uart: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\1-uart\__init__.py", line 40, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\1-uart\pd.py", line 22, in <module>
    from common.srdhelper import bitpack
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder dcf77: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\dcf77\__init__.py", line 28, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\dcf77\pd.py", line 22, in <module>
    from common.srdhelper import bcd2int
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder ds1307: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\ds1307\__init__.py", line 25, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\ds1307\pd.py", line 23, in <module>
    from common.srdhelper import bcd2int
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder graycode: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\graycode\__init__.py", line 24, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\graycode\pd.py", line 24, in <module>
    from common.srdhelper import bitpack, bitunpack
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder jtag_ejtag: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\jtag_ejtag\__init__.py", line 25, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\jtag_ejtag\pd.py", line 21, in <module>
    from common.srdhelper import bin2int
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder ook_vis: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\ook_vis\__init__.py", line 25, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\ook_vis\pd.py", line 21, in <module>
    from common.srdhelper import bcd2int
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder parallel: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\parallel\__init__.py", line 34, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\parallel\pd.py", line 21, in <module>
    from common.srdhelper import bitpack
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder rtc8564: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\rtc8564\__init__.py", line 25, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\rtc8564\pd.py", line 21, in <module>
    from common.srdhelper import bcd2int
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder sdcard_sd: import by name failed:ModuleNotFoundError:No module named 'common.srdhelper'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\sdcard_sd\__init__.py", line 24, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\sdcard_sd\pd.py", line 21, in <module>
    from common.srdhelper import SrdIntEnum, SrdStrEnum
ModuleNotFoundError: No module named 'common.srdhelper'

srd: Failed to load decoder sdcard_spi: import by name failed:ModuleNotFoundError:No module named 'common.sdcard'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\sdcard_spi\__init__.py", line 68, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\sdcard_spi\pd.py", line 21, in <module>
    from common.sdcard import (cmd_names, acmd_names)
ModuleNotFoundError: No module named 'common.sdcard'

srd: Failed to load decoder xfp: import by name failed:ModuleNotFoundError:No module named 'common.plugtrx'.
srd: Traceback (most recent call last):
  File "C:/Program Files/DSView/decoders\xfp\__init__.py", line 38, in <module>
    from .pd import Decoder
  File "C:/Program Files/DSView/decoders\xfp\pd.py", line 21, in <module>
    from common.plugtrx import (MODULE_ID, ALARM_THRESHOLDS, AD_READOUTS, GCS_BITS,
ModuleNotFoundError: No module named 'common.plugtrx'

Protocol decoder enabled!

Hotplug thread start!

问题找到了，原因是windows10安装Python的时候勾选了添加到环境变量，在系统环境变量里面会存在一个PYTHONHOME的一个变量名
[image]
导致程序运行时模块读取不到
[image]

解决方式：删除PYTHONHOME变量，重新打开DSView