DreamSourceLab/DSView #477: I3C协议支持

wohaoa, 2022-03-08:
目前手机平台都要求支持I3C，目前传感器都已经开始支持I3C(加速度，陀螺仪，地磁等等)，请问能添加I3C协议么？

wohaoa, 2022-03-08:
[image]

DreamSourceLab, 2022-03-11:
@wohaoa 
Could you send us the I3C protocol specification?
Thanks.

wohaoa, 2022-03-14:
@DreamSourceLab 
The following is mipi_I3C-Basic_specification_v1-1-1
[mipi_I3C-Basic_specification_v1-1-1.pdf](https://github.com/DreamSourceLab/DSView/files/8243006/mipi_I3C-Basic_specification_v1-1-1.pdf)

wohaoa, 2022-03-21:
@DreamSourceLab 
Does the mipi_I3C-Basic_specification_v1-1-1.pdf I sent meet your requirements, which contains the I3C protocol specification?

DreamSourceLab, 2022-03-31:
@wohaoa 
Yes, got it.
Thanks.

wohaoa, 2022-04-07:
@DreamSourceLab When is it expected to support this protocol? thank you.

mythbuster5, 2024-02-09:
@DreamSourceLab It has been 2 years since the issue came out. Is there any progress? Do you can any specific plan to release it?

strongtek, 2024-04-03:
co-ask, please add i3c support.

RaboLab, 2024-07-15:
@DreamSourceLab It has been 2 years since the issue came out. Is there any progress? Do you can any specific plan to release it?

ChiaoY, 2024-08-08:
@DreamSourceLab It has been 2 years since the issue came out. Is there any progress? Do you can any specific plan to release it?

mythbuster5, 2025-03-24:
@DreamSourceLab Could you please take a look? Thanks!

xyphro, 2025-07-24:
Try this out - made it for sig rok but it should be compatible. Curioud too if it runs with dsview:

https://github.com/xyphro/Sigrok-I3C-decoder

Posted it to dsview guys several months ago.