DreamSourceLab/DSView #478: DSView behavior in Repetitive mode

echelon2020, 2022-03-10:
Hi All

I am new to DSView and also DSLogic UPro 16.

What I don´t understand so far

If I use repetitive mode I would think this should work like a scope in "Run" mode with a certain trigger.
Meaning the analyzer wants for triggers and captures and displays after that trigger.
What I see is that it also capture after the repetivie time always even there is no trigger event. This makes this mode for me useless because most of the time there is no useful information on my screen as I see signals which were not triggered. Do I do something wrong or is this by design ?

regards
Uwe

DreamSourceLab, 2022-03-11:
@echelon2020 
Trigger is decided by your settings.
If you don't set any trigger conditions, the capture will restart automatically.
If you setup some trigger conditions, the caputre will wait for the trigger.

echelon2020, 2022-03-11:
Thank you. This was my expectation. 
i set an esge trigger on one channel. Single mode waits forever for the trigger. When edge occurs triggers and display

In repetitive mode also waits for the first trigger and displays signals but then after the wait time overwrites the display with whatever happens but there was no edge…

this happens in streaming and in buffer mode….