DreamSourceLab/DSView #481: Display Error When Decoding I2C/IIC(1/2)

LightgardenCC, 2022-03-26:
`
[image]
`

LightgardenCC, 2022-03-28:
[image]
`Start` / `Stop` decoding flags has zoom problem in Arch-Linux， and sometimes I can't even see these flags while other flags are normal.
> Although it's not a large problem, it can be not a little friendly

DreamSourceLab, 2022-03-29:
@LightgardenCC 
Please upload the *.dsl file for check.
Thanks.

LightgardenCC, 2022-04-06:
[DSLogic-la-220406-213801.txt](https://github.com/DreamSourceLab/DSView/files/8427342/DSLogic-la-220406-213801.txt)

> I use *.txt instead of *.dsl, because GitHub does not support DSL file format.

dreamsource-tai, 2022-04-07:
@LightgardenCC 
The uploaded file have not attach any decode protocol, we do not known how to setting channels,please upload it again!

LightgardenCC, 2022-04-07:
[DSLogic-la-220407-133248.dsl.txt](https://github.com/DreamSourceLab/DSView/files/8441555/DSLogic-la-220407-133248.dsl.txt)
[image]

Protocol -> 1: I2C
SCL -> CH2
SDA -> CH3

> As shown in the figure above, there is a fault with the `start` flag.

BrainsLab, 2022-05-20:
Hi! I'm getting the same problem with displaying S and P spots, but with additionally errorneous decoding. My device reproducing the same repeated test I2C frame:  addr 0x40 write, followed by sequence of four bytes: 0x01, 0x02, 0x03, 0x04. It can be seen in attached file that the first frame was decoded properly, but the next frames got errors - there are 7 dots in addr decode (1759.26 us), then 7 dots in the last byte 0x04 (2469,42us) and undecoded two bit intervals before the STOP. It's very annoying and I can't do my work with this instrument. Please fix it ASAP. The file attached was renamed to txt.
[DSLogic-la-220521-001220 ERRORNEOUS I2C DECODING.txt](https://github.com/DreamSourceLab/DSView/files/8744698/DSLogic-la-220521-001220.ERRORNEOUS.I2C.DECODING.txt)