DreamSourceLab/DSView #484: DSView V1.12 x64 crashes, frozing and get slow with lot SPI data in stream mode

rob-bits, 2022-04-04:
I have configured the following setup:

- DSLogic Basic connected to a Win10 x64 machine
- 50 MHz samplerate is set
- Stream mode, sampling 6 channels
- 0-1 channel set to a slow GPIO period of ~3.2ms
- 2-5 channel configured to SPI (MISO, MOSI, CLK, CS)
- MISO and MOSI lines have decoder
- CLK frequency is ~5MHz, clock never stops
- CS frequency is ~160 kHz
- 16 bit data read/written
- time frame is 2 second (~320 000 data is displayed on MOSI line and the same on MISO)

Starting the measurements works fine. After couple of measurements it randomly crashed.  Giving errors like this:
[image]

Sometimes the program becomes unstable probably because of the data amount. 

I dont know what is the problem. I just regularly restarting the application but it is really annoying.

dreamsource-tai, 2022-04-30:
@rob-bits 
Please download the new version to experience.
url:  https://www.dreamsourcelab.cn/software/DSView_v1.2.0rc6_x64_setup.exe

rob-bits, 2022-05-04:
@dreamsource-tai 
Thanks, soon I am testing and get back to this topic.