DreamSourceLab/DSView #485: Consolidate lib*4DSL compilation steps into the main build

anatol, 2022-04-14:
Currently, `dsview` project build requires building and **installing** 2 private dependencies (libsigrok4DSL and libsigrokdecode4DSL).

I am a packager of dsview for Arch Linux. And for the packages, the current dsview structure requires adding two extra OS packages. There seems no way to build dsview without install *DSL packages first.

But if the `*DSL` packages are considered private to DSView then distributing them as separate packages is considered a bad practice. Please make it possible to build and install all required libraries as a single package **without** installing *DSL packages first.

anatol, 2022-04-14:
But much better if all the libsigrok4DSL and libsigrokdecode4DSL land upstream. This way you don't need to keep the fork of these projects in your codebase.

dreamsource-tai, 2022-04-15:
@anatol 
libsigrok4DSL and libsigrokdecode4DSL we need customized development。In the future, we will integrate them into a project for compile.