DreamSourceLab/DSView #486: Dose DSView support display data with ascii format for all the communication protocol?

CV-Bowen, 2022-04-23:
I think this is a very useful function, and the uart has supported. But spi, i2c etc are not supported.
Do you have a plan to do this? Maybe add a High-level protocols for transfer hex data to ascii char is a easy way?

dreamsource-tai, 2022-04-24:
@CV-Bowen 
What is your operating system? if your system is linux, you can pull the project, switch to "dev-tai" branch, and compile it.
we already support this feature.we will release a new version this week for windows.

CV-Bowen, 2022-04-24:
@dreamsource-tai My operating system is Window. I'm very happy to hear that. I will wait for your next version for Windows.

dreamsource-tai, 2022-04-30:
@CV-Bowen 
Please download the new version to experience.
url:  https://www.dreamsourcelab.cn/software/DSView_v1.2.0rc6_x64_setup.exe

closed 2022-06-06 by CV-Bowen