DreamSourceLab/DSView #487: crash in ubuntu22.04

EGQM, 2022-04-25:
the terminal output:
```
srd: Only PD API version 3 is supported, decoder eeprom93cxx has version 2:unknown error.
srd: Failed to load decoder eeprom93cxx: API version mismatch
srd: Failed to load decoder ir_sirc: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

srd: Failed to load decoder sle44xx: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

srd: Only PD API version 3 is supported, decoder maxim_ds28ea00 has version 2:unknown error.
srd: Failed to load decoder maxim_ds28ea00: API version mismatch
srd: Failed to load decoder ir_sirc: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/ir_sirc/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

srd: Failed to load decoder sle44xx: import by name failed:ImportError:cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py).
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/__init__.py", line 26, in <module>
    from .pd import Decoder
  File "/usr/local/share/libsigrokdecode4DSL/decoders/sle44xx/pd.py", line 20, in <module>
    from common.srdhelper import bitpack_lsb
ImportError: cannot import name 'bitpack_lsb' from 'common.srdhelper' (/usr/local/share/libsigrokdecode4DSL/decoders/common/srdhelper/__init__.py)

[1]    21959 segmentation fault (core dumped)  DSView

```

dreamsource-tai, 2022-04-26:
@EGQM 
Did you add the eeprom93cxx protocol yourself?Please upload it for check.

dreamsource-tai, 2022-04-26:
@EGQM 
你现在应该是在中国吧？如果是这样的话，你可以在微信公众号里输入“梦源科技”，然后关注一下我们的官方公众号，在上边跟我们的技术客服进一步沟通。谢谢！

EGQM, 2022-04-26:
@dreamsource-tai 我加了你们的公众号。我没有加入eeprom93cxx协议，就是直接clone dsview的仓库，然后编译安装的。能编译和安装，但是不能运行。之前ubuntu21.10中没有问题，但是升级到ubuntu22.04后，异常。即使重新编译安装后，依旧异常。

EGQM, 2022-04-26:
@dreamsource-tai 最新的git版本和最后的release版本都尝试了，都没成功。

dreamsource-tai, 2022-04-26:
@EGQM 
你在公众号跟客服说清楚，说明对接到开发这边来，然后我们远程看看是什么原因。

EGQM, 2022-04-27:
@dreamsource-tai 微信已经申请加了你的技术，但是一直未通过。

dreamsource-tai, 2022-04-28:
@EGQM 
你关注公众号后，联系客服，然后我们进一步沟通，比如远程操作。不用加微信啊。

EGQM, 2022-04-28:
@dreamsource-tai 已经联系客服，客服给我一个你们的周姓工程师的微信让我加，我申请了，但是他一直没有通过。

EGQM, 2022-04-28:
@dreamsource-tai 根技术沟通，已经解决了。
删除旧的解码器即可：
```
sudo rm -rf /usr/local/share/libsigrokdecode4DSL/decoders/eeprom93cxx
 sudo rm -rf /usr/local/share/libsigrokdecode4DSL/decoders/maxim_ds28ea00/
```

closed 2022-04-28 by EGQM