DreamSourceLab/DSView #488: Continous real time mode?

mateusz-szczyrzyca, 2022-04-28:
Hello,

I have one problem with DSView or maybe it's a feature, I don't know.

Is it possible to run in continous real time mode like pulseview? Because every single mode always shows me this:

[image]

And I can't see the data till reading stops. It's frustrating.

I just want to see channels like in Logic2 or Pulseview - constant line which shows me high or low state, depends on situation.

I've read manual and some docs but I can't see how to set this.

dreamsource-tai, 2022-04-30:
@mateusz-szczyrzyca 
Now the image is displayed only after the acquisition is completed. In the future, we will change to display while acquiring.

mateusz-szczyrzyca, 2022-04-30:
@dreamsource-tai is any ETA about this change?

dreamsource-tai, 2022-04-30:
@mateusz-szczyrzyca 
就是说这个时实显示波形的功能，是我们后边的一个开发目标，只是还需要阁下等等。

closed 2022-05-01 by mateusz-szczyrzyca